FrSky-OS/HORUS-X10-X10S-X12S-bug-report #118: 1.3.01 L9R mode, SWR always zero, and no warning when removing the antenna

OlivierC-FR, 2017-06-16:
I get no SWR reading when the radio is in L9R mode (with option Antenna = outside), and if I remove the external antenna it stays at 0 and also I do not get any "no antenna" warning.

Also I'm beginning to suspect the radio is maybe only using the internal antenna in this mode, because when I fly my plane at 3-4km range, with a 2.4Ghz patch antenna on the external SMA, and I rotate/track/aim at the plane, the RSSI displayed in my FPV goggles does not seem to react accordingly.
Plus the system is struggling to go beyond 4-5kms, with a L9R receivers there's many reports of pilots going way beyond that, with Taranis mostly.

In D16 mode it all works as expected, I get a SWR value, and warnings when the antenna is not plugged in.

Bug replicated on Open TX 2.2 on the RCgroups forum

janekx, 2017-06-19:
what OS version please?

janekx, 2017-06-19:
1.2.25 same issue in LR12

OlivierC-FR, 2017-06-19:
OS is FrSky-OS 1.3.01 (international), the radio is an euro version.
And I fixed the title, of course the mode is L12R, and L9R is the receiver :)

OlivierC-FR, 2017-06-19:
Made more tests last evening, with my plane set to fly autonomoulsy in circles at 3000 meters, orienting the external patch antenna made no difference with RSSI (seen on the OSD, feed by the L9R), while I managed to drop the RSSI a lot by putting my hand in front of where the internal antenna is. So I'm like 50% certain the outside antenna is not emitting in this mode.

Webbsolution, 2017-06-23:
Oliver - I tested my two radios with NON Eu firmware on OTX 2.2 - I believe the antennae is transmitting just fine  - I used a basic RF analyzer and coupled it directly to the external antennae. The coupler is RF shielded so I basically only have input from the external  output connector.  I tested d16 and L12 modes and then created baseline with the radio off as well - its easy to see when the external antennae is transmitting. 

check out the screen caps on this thread - https://www.rcgroups.com/forums/showpost.php?p=37780274&postcount=10679  

I would start investigating your aircraft's local RF sources or that antennae itself as your range issue - as you know FPV installs are typically challenged by local Video TX antennae etc. Since its quite clear that we have a bug that killed the RAS meter ...you are blind as to weather or not the TX is happy with that antennae.

Webbsolution, 2017-06-24:
So the bug that is inarguable is that the RAS number read out in L12 mode is BROKEN - the external Antennae alarm is also BROKEN.  IF you wish to reproduce the behavior - bind a L9R in L12 mode - switch to internal + external antennae - then remove the external antennae - Compare your experience when doing the same test in d16 mode. There should be no additional data needed now - its a bug guys.

Adelaxie, 2017-08-21:
yes,it is. we disabled all data (SWR+sensor data)display when no telemetry received .L9r is a single-way receiver will has no data received always.
So....
Will modify the behavior.

Adelaxie, 2017-10-30:
Will fix it in next release.

Adelaxie, 2017-11-20:
Please check next release

closed 2017-11-20 by Adelaxie