FrSky-OS/HORUS-X10-X10S-X12S-bug-report #119: Intermittent Telemetry Lost alert when using IXJT and XJT

MIKE-DAILEY, 2017-06-16:
When using the IXJT with telemetry enabled X receiver and a second X receiver (telemetry disabled) with a XJT module (with telemetry disabled) at the same time the system intermittently reports “Telemetry Lost”.

RealTadango, 2017-06-16:
See also https://www.rcgroups.com/forums/showthread.php?2827965-FrSky-%28Horus%29-dual-RF-%28IXJT-and-XJT%29-issue-thread-%28feedback-needed%29

MIKE-DAILEY, 2017-06-16:
Thanks for adding the RCG link.

MIKE-DAILEY, 2017-09-01:
@Adelaxie does this item "5. Add external RF module S.Port switch, to avoid data conflicts on RF System interface;" in 1.4.02 have any impact on the problem with using the XJT at the same time as the IXJT?

RealTadango, 2017-09-01:
I think it puts the XJT in both switches up position only.

Adelaxie, 2017-11-23:
Issues out of date

RealTadango, 2017-11-23:
So... it won't be fixed?

Adelaxie, 2017-11-24:
It is because that you did not power off the S.PORT of internal module(not same with telemetry off).
Anything  confused you please post here.

MIKE-DAILEY, 2017-11-24:
Not sure I understand.  On the RF SYSTEM screen I see control of S.P. MODE for the external module.  How do I control S.Port on the internal IXJT module?

Adelaxie, 2017-11-24:
oh, wrong description. it should disable the external s.port/ in word , two module with s.port enabled will make trouble. even the telemetry of receiver is off.
But Now R9M also need to do it like this and we are discussing that if the logic can be more clear and easy.
Will see.

RealTadango, 2017-11-24:
This issue is NOT about the SPORT not being disabled on the external module. As i have explained before, and in email, this issue is about the external module transmitting while the internal module is receiving. The external module RF output is flooding the internal module which cannot receive the telemetry data properly. This works fine in close range (although telemetry packet loss is higher already then with a single module) but at a range of +/-100 telemetry reception becomes unusable and will dropout often. Using the external module in range test mode works perfectly because the output power is much less.

Solution: The external module should syn the RF packets so it won't transmit while the internal module is receiving. Maybe the heartbeat signal can be used for that. A simple firmware update for the XJT is enough. Please contact me for details and testing assistance.