FrSky-OS/HORUS-X10-X10S-X12S-bug-report #120: Create or modify FrSky Horus X12S firmware?

ockdavid, 2017-07-26:
I'm looking for a way to edit or create my own firmware and install it on my Horus X12S, is that possible? I'm reading some things about OpenTX Companion but I'm not finding what I want. Thanks for your help!

strgaltdel, 2017-07-27:
Ich kehre zurück am 13.08.2017.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "[FrSky-OS/X12S-bug-report] Create
or modify FrSky Horus X12S firmware? (#120)" gesendet am 27.07.2017 00:48:30.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

janekx, 2017-07-27:
Deleted email autoanswer comment ....

In case of FrSky-OS it is not possible as it is under FrSky dev. OpenTX is open source, you can make the clone....

strgaltdel, 2017-07-27:
Ich kehre zurück am 13.08.2017.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re: [FrSky-OS/X12S-bug-report]
Create or modify FrSky Horus X12S firmware? (#120)" gesendet am 27.07.2017 09:54:26.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

closed 2017-08-15 by ockdavid