FrSky-OS/HORUS-X10-X10S-X12S-bug-report #121: I don't understand what is CURVE LIBRARY

ockdavid, 2017-08-15:
Hello! I'm working with the Horus X12S and in the MDL menu there is the CURVE LIB option that is Curve Library. However, I don't understand what the Horus User Guide try to explain. It says "Here the user can create custom curves that can be used via "MIXER", the operation of this meu is similar to setting up "EXPO" and "MIXER". That's all. Also I don't know what is Mixer yet. So I would appreciate your help and uncomplicated explanations! Thanks.

MIKE-DAILEY, 2017-08-16:
The EX MIXER provides the ability to do custom mixes if the pre-defined mixes cannot do the mix you want.  The curve lib is sometimes used with the EX MIXER to apply a custom curve to a mix. Once the custom curve is in the LIB you can use it as needed.

MIKE-DAILEY, 2017-08-21:
Glad it helped.

closed 2017-08-21 by MIKE-DAILEY