FrSky-OS/HORUS-X10-X10S-X12S-bug-report #122: No vario tones being played.

radfordc, 2017-08-17:
I am using an FrSky vario and I can't get the Tx to play the vario tones that indicate rising or descending.  The vario does work, as I have the Tx set to announce the altitude every 10s and this works correctly.  However, I can't get the vario tones to work.

The receiver is on and the vario data is active on the screen.  In the Special Function menu I have a line set as follows: the en box is checked, selected the "once" choice, set switch as "SWA down", selected the "PlayVario" choice, the last field is blank. 

When SWA is down the vario tone beeps should start, but I'm hearing nothing.  This problem has existed since I bought the Tx last year.  I updated to the latest 1.4.02 firmware and it still doesn't work.

MIKE-DAILEY, 2017-08-17:
@Adelaxie Some background on this problem.  This feature seems to work on some systems, but not others like the system the @radfordc has. I works fine on two of my production systems and I'm wondering if there are no beeps because of missing sound files.  If we knew the sound file names used we could check the system @radfordc has.

Mike

Adelaxie, 2017-08-18:
@radfordc @MIKE-DAILEY 
Now the FrTX supports this function, we used the beep genrated inside not from any files.
So keep you did not disable the beep sound.

MIKE-DAILEY, 2017-08-18:
Thanks @Adelaxie that gives @radfordc some things to check on his system.

closed 2017-11-23 by Adelaxie