FrSky-OS/HORUS-X10-X10S-X12S-bug-report #123: Timer as LSW

Nesan65, 2017-08-18:
1.4.02(Beta)
HI I Try to use the timer as trigger to a function.  but it seems not working. could some one check it please.
if it work can we use it as sequencer?

MIKE-DAILEY, 2017-08-18:
It looks like the CONST value cannot determine the timer value.   Is the CONST SRC 2 seconds, min, hours?

Mike

Nesan65, 2017-08-18:
even I tried like "Timer1 > Timer2"  not working.
but "Timer1=Timer2" is ok ???@#*~*

Adelaxie, 2017-08-21:
@Nesan65  We modified it months ago, do you mean it still doesnot work? we tested it ok.

MIKE-DAILEY, 2017-08-21:
How does the CONST value relate to the timer.  Is the CONST number seconds?

Nesan65, 2017-08-21:
I think it is not working or may be I do not configure properly.! in this case please let me know one sample please.

Nesan65, 2017-08-24:
can anybody explain how to do it Timer as LSW Please. I could not do it!

MIKE-DAILEY, 2017-08-24:
I guess I don't understand how the feature works.  I thought you could use a logic switch that would go true when the counter went over or under a counter number set in CONST, but I'm not able to make it work.

MIKE-DAILEY, 2017-09-02:
Tested in the next firmware to be released.  The CONST is in seconds.  So if the timer starts a 2 min and you want the LS to be true on count down to 1 min the CONST is set at 60.

Nesan65, 2017-09-03:
So it is not working in F.W 1.4.02(Beta). and do could you say approx when they release it?
Any way thank you for let me know about this.

MIKE-DAILEY, 2017-09-03:
It would not work for me in 1.4.02, but does in the next beta to be released.  Don't know when it will be released.

closed 2017-10-27 by Adelaxie