FrSky-OS/HORUS-X10-X10S-X12S-bug-report #124: Joystick, logical switch and play track with special function

goryc, 2017-08-31:
With the new firmware 1.4.02 does not work anymore, first worked perfectly!!!

[image]

 See the last row LSW2
[image]

MIKE-DAILEY, 2017-08-31:
I tried a simple logic switch to trigger on J1 value being over or under a CONST and the logic switch does not change state for me as well.

Adelaxie, 2017-09-01:
yes,we are testing the fixed firmware. thanks for feedback. please pay attention on update on website.

Nesan65, 2017-09-05:
i make it work by assign a channel to M-Y and M-X in EXMixer then call it from LSW it works. and this way two channel can be control by M-X and M-Y. like camera axil

Adelaxie, 2017-10-27:
fixed on v1.4.3

closed 2017-10-27 by Adelaxie