FrSky-OS/HORUS-X10-X10S-X12S-bug-report #127: Overlay and icons missing

alex998r, 2017-10-09:
Upgraded to the latest 1.4.02 firmware, in the main screen the right top icons are missing ( battery, Rf )
In the second screen the whole overlay is missing
[image]
[image]

andymeade, 2017-10-09:
Confirmed here as well - quite annoying when flying as I have no battery level indicator!

MIKE-DAILEY, 2017-10-09:
You did not copy all the files provided correctly to the flash memory drive. See steps here https://www.rcgroups.com/forums/showpost.php?p=38101065&postcount=11471

alex998r, 2017-10-09:
Ok Mike, thanks for the info, now it works

closed 2017-10-09 by alex998r