FrSky-OS/HORUS-X10-X10S-X12S-bug-report #128: More Dual rates on each  flight mode/ Throttle in Glider Model

cpg007, 2017-10-13:
When you select a Glider model you only can have one Dual Rate for channel on each flight mode.
Could be posible to have more rates assigned to one switch?. You have this but only if Model is an Airplane.
Could be posible have the throttle channel in Glider model and not only Motor function. Many of us fly Electric glider whith throttle channel in J3.

Adelaxie, 2017-10-16:
@cpg007 

> Why is not posible to use de same Dual Rate/exp of model Airplanes in glider Models? 

Based on now structure, it is not possible.

>Could be posible have the throttle channel in Glider model and not only Motor function. Many of us fly Electric glider whith throttle channel in J3.

You can set a AUX channel with source with J3,but be careful that there is no protection fuction to the channel(no throttle hold / throttle cut).

Boroal, 2017-10-16:
Adelaxie, the lack of throttle hold on an AUX Channel is a problem I came across when using it to set up differential thrust on a twin motor plane. I have one motor that is always “live”. It really does need the ability to have some form of throttle hold on the AUX when you are running two or more motors on separate channels. The use of differential thrust on these planes is quite common.

cpg007, 2017-10-16:
I set the Aux channel as J3 and I made a throttle cut with ExMixer. 
When a Switch is ON it replace Aux channel (J3) with a Curve to -100.
ATTENTION: It is dangerous, sometimes it works and sometimes not.