FrSky-OS/HORUS-X10-X10S-X12S-bug-report #129: Exmixer vs Mixer

cpg007, 2017-10-16:
Why can not "ExMixer" and "Mixer" have the same functionality?
With "Exmixer" I can not concatenate several mixers with +=, *= or :=.
I can only do it in Custom with "Mixer". But in Custom I do not have the predefined mixers.
If I define a model other than Custom, I always reach a dead-end street.
To do what I need I always have to use a Custom Model, it's like using OpenTX but without all its power.

Regards 
Carlos Pelayo

Adelaxie, 2017-11-24:
>With "Exmixer" I can not concatenate several mixers with +=, *= or :=.

There is 10 slot to do the mixers/

cpg007, 2017-11-24:
Hello Adelaxie,
In the Extra Mixer of Airplane model (version 1.4.04)  I press OK and the slot  only pop up Edit and Reset but no Create. 
What I have to do to get in Extra Mixer Pop up EDIT, RESET and CREATE?
Is it possible that I am doing something wrong?  

Regards
Carlos