FrSky-OS/HORUS-X10-X10S-X12S-bug-report #131: 1.4.0.2 Rates and EXPO using flight modes vs switch

uragano47812, 2017-10-23:
I see a strange behaviour when i setup a new model. In this case i first created 3 flight modes and assigned them to a 3-pos switch. Then when i went to Rate menu at first page i have AIL so i put my EXPO and Weight, 3 values (1-2-3) and switched from "Switch" to "flightmode" and it works for AIL as it should.
Then i selected ELE and i see it was already with flightmode selected so i put my 3- different values for expo and weight and so i did for RUD. BUT i noticed that at this point only AIL was working changing rates per flightmode selected but the other surfaces not.

After some more test i see that even if i selected flightmode at first for AIL and it was already selected for the other surfaces i had to select for ELE switch and the reselect flightmode to be able to working as aspected and also for RUD i have to do the same thing.

I hope this will be fixed on other FrOS release.

MIKE-DAILEY, 2017-10-23:
@Adelaxie verified that the same problem exist in 1.4.03. 

Mike

Adelaxie, 2017-10-26:
Thanks for all your help. we are fixing it.

Adelaxie, 2017-10-27:
Please pay attention on new firmware next week updated on website. @uragano47812

closed 2017-11-20 by Adelaxie