FrSky-OS/HORUS-X10-X10S-X12S-bug-report #132: Flaps function

craigblk, 2017-10-24:
Could anyone help me get flaps setup on my X12S, pulling my hair out trying to figure it out.

Nesan65, 2017-10-27:
i assigned channel 6 and 7 for flaps  "two servo"  set SWC three position flap no flap - takeoff flap - landing flap for my Top flight Corsire 
[image]
[image]
[image]

Woita2095, 2018-02-01:
And what about the slowing-down the moving of the flaps?
And compensation by the elevator?
(See my Issue #174)