FrSky-OS/HORUS-X10-X10S-X12S-bug-report #134: Throttle warning does not work on custom model

Drashiel, 2017-10-27:
Hi,
I have a custom model on my Horus running 1.3.01. When I boot up radio with throttle stick at not minimal and this custom model selected, I don´t have throttle warning. Nor can add on warning setup page.

I think it could be great have this feature since it could be disabled on warning setup.

Adelaxie, 2017-10-30:
1. Throttle channel can be assigned in custom mode.
2. Throttle channel name can be modified under custom mode.
3. Throttle warning is based on the value of the phisical stick position.

So, very complex if the warning added.

Drashiel, 2017-10-30:
You can add a channel selection on switch warning setup to select what channel throttle is and add position read as rest of models.

Adelaxie, 2017-10-31:
yes, if want to add , then need customers to set it manually/.

Drashiel, 2017-10-31:
So, it is possible. Maybe you can consider add on next firmware release?

Adelaxie, 2017-11-01:
Will suggest and if any progress will post it here. Thanks for sugegstion.

Adelaxie, 2017-11-20:
supported in next release v1.2.01 (v1.5.01 for x12s)

closed 2017-11-20 by Adelaxie