FrSky-OS/HORUS-X10-X10S-X12S-bug-report #135: X10 Horus does not work with irangex multi module

Crunch69, 2017-11-02:
X10 Horus does not work with irangex multi module Works in x7 not Horus this is really important since opentx will not be completed until sometime in 2018 making my radio just a paper weight

Crunch69, 2017-11-02:
Fros needs a computer interface like open rx so you can build models easily

gumush, 2017-11-08:
Same as mine , i guess opentx will support that but Fros needs to update for IrangeX support.

Crunch69, 2017-11-09:
Multi module mode should work with all the modules see Lin to opentx https://github.com/opentx/opentx/search?utf8=✓&q=multi+module&type=

MIKE-DAILEY, 2017-11-09:
With FrOS I use the newest 4 in 1 switch select module that supports all the protocols.  It also has a serial switch setting that makes it compatible with the OpenTX multimode.  Works well, but of course when running FrOS the protocol must be changed with the module rotary switch.  I use the iRangeX on my FrSky system that only run OpenTX.

Crunch69, 2017-11-20:
What we need is serial support like it works in thx 9 and x7 multimode

martinvanmiddelkoop, 2020-09-05:
> With FrOS I use the newest 4 in 1 switch select module that supports all the protocols. Works well, but of course when running FrOS the protocol must be changed with the module rotary switch. 

@MIKE-DAILEY - would you give me some tips on n how to setup the IRX4+ on FrOS perhaps? I don’t get it yet. Am using the most recent firmware of FrOS

MIKE-DAILEY, 2020-09-05:
In FrOS select the internal module OFF and external module ON.  On the FrOS external module settings select PPM.

martinvanmiddelkoop, 2020-09-05:
Thanks for your very fast reply, @MIKE-DAILEY !

I am afraid that the latest version of FrOS has disabled this option. I can only select modes XIT, R9M and R9M ACCESS.

MIKE-DAILEY, 2020-09-05:
In that case the module cannot be used.

martinvanmiddelkoop, 2020-09-05:
O, actually I am using the EU LBT version of the X10S, but probably the same will happen with the FCC version

I might do a test with OpenTX instead, that should work, it seems.