FrSky-OS/HORUS-X10-X10S-X12S-bug-report #140: [Enhancement] drivers to allow x12s radio to be used as a joystick for flight stimulator

dontascii, 2017-11-19:
OpenTX has a build that allows you to use the radio as a input for flight simulators instead of the mass storage driver.  This is the only feature that would entice me to use OPEN TX over FrOS. Are there any plans to develop an input driver so usb can be used as a joystick?
Thanks in advance ! I love frOS!

Adelaxie, 2017-11-20:
Dumplicate with #115 please check.
Will do it , thanks for feedback.
:)

dontascii, 2017-11-20:
This is not the same request.  #115 was for the ability to use the gimbals to drive the menu system.  That is not what I requested above.

Adelaxie, 2017-11-20:
oh ,sorry,but this feature we have received some suggestions and its already in our todo list.
Thanks again.

dontascii, 2017-11-20:
That's excellent!  If both mass storage driver and usb game controller option work on the same firmware version, that would be neat. OpenTX can't do that.

kilrah, 2017-11-20:
> OpenTX can't do that.

It can in 2.2.1 :)