FrSky-OS/HORUS-X10-X10S-X12S-bug-report #160: [enhancement] More PITCH curve points in Helicopter Model

jperez-murcia, 2017-12-26:
For finest and better helicopter PITCH curves setup, much more than 7 points are needed. For F3C (helicopter acrobatics) setups, as more points, a better and smooth setup will be performed. Seven points aren't enough.

OpenTX curve handling would be great in FrOS. A lot of curve points and full freedom to configure such points,

Please, consider this enhancement for future releases because; seven point are enough for sport flying, but not for competition.

Great work ...

Adelaxie, 2017-12-27:
@jperez-murcia  Now the point can support up to 22. what version do you use?

jperez-murcia, 2017-12-27:
Hi Adela...

AFAIK I'm using, last version? ,... 1.4.04 .... but I can't configure over 7 points ... may be I'm missing something?

regards

Adelaxie, 2017-12-28:
Please select MULTI_LINE or MULTI_SMOOTH mode of the dropdown widget in the upper middle of the page.

MIKE-DAILEY, 2017-12-29:
@Adelaxie I think the MULTI_LINE or MULTI_SMOOTH function might be broken.  You can set 21 points but the points will not scroll to reach past P7.

Adelaxie, 2018-01-30:
@jperez-murcia  any problems still? If No more question please close the issue.

jperez-murcia, 2018-01-31:
my apologies .... no additional problems.

closed 2018-01-31 by jperez-murcia