FrSky-OS/HORUS-X10-X10S-X12S-bug-report #161: Issue with Flight Modes and Pitch curves Horus X10S Firmware 1.2.02

Maverick440, 2018-01-03:
When in Helicopter mode (HR3) you cannot assign pitch curves to ether a switch or flight modes.
When curves are asigned to a flight mode swithch, switch activation has no affect on the cyclic servos (SwashPlate)
If you dont use flight mode and try to assign the pitch curves to a switch, activation of that swith only affects one of the cyclic servos and not all three (only activates the AlL channel)

Maverick440, 2018-01-03:
Ok so done some more testing today and teh issue im having appears to be down to the setting in SWASH RING, I had the swash ring set to:
Rate =100%
Ale   = +50
Elv    = +50
Pitch = +60
and this seams to be the issue, if any of the swash ring is "out of ballance" "Flight mode" and "Switch selection" wont work within Pitch curves. As soon as you ballance the cyclic outputs it at apears to work OK.
Ill do some more setups tommorow and confirm my findings.

Adelaxie, 2018-01-04:
@Maverick440  Where you downloaded the config file? we can not reproduce it.
we assumed that the wrong config file you used ,maybe?

Maverick440, 2018-01-04:
I downloaded  the update file from https://www.frsky-rc.com/x10s/  and as I have the EU version i used the file X10_mode2_EU_1202_frtx and the followed the FrOS firmware updating instructions.

Maverick440, 2018-01-04:
OK so this issue is as follows, im trying to setup a Flybar helicopter with a tail only Gyro.
The gyro has two modes which are switchable, Heading Hold, and Rate mode.
You can set up Pitch curves and trottle curves and assign them to a swich under Flight Mode (IDLE1,2&3) and all works great however
when I try to set the gyro, this is causing the issue.
I have Channel 5 set to Gyro (single gyro attached)
I now go into the Gyro screen.
In order to set the gyro up i need to be able to switch between Rate +00 to +49 (Heading hold)
and -00 to -49 (Rate mode) the only options you are given below the NORMAL tab are
IDLE1, IDLE2, IDLE3, HOLD, F.Mode6, F.Mode7, F.Mode8 and F.Mode9.
So i select F.Mode6 for Rate +00 to +49 and F.Mode7 for rate -00 to -49
Go back into FLIGHT MODE and assign a switch to F.Mode6&7 
F.Mode 6 = SWFLP_1
F.Mode 7 = SWFLP-2
Once you do this I can setup the gyro and switch in flight between Heading hold and Rate mode but you cant switch between IDLE 1,2 &3 because the Gyro (assigned to Flight mode 6&7) overide Flight modes 1to5 which by default are IDLE1,2,3 & Hold.
So how can i set up the Horas so that i can set Pitch and Throttle curves to a single switch and Gyro to a switch so that i canswitch beween IDLE 1,2 and 3 as well as swich between Heading hold and rate mode on the gyro.
If i try and set the pitch curves to a switch insted of a flight mode it only switches channel 2

Maverick440, 2018-01-04:
OK ive got a handle on the logic and have it all sorted i think.
Set throttle and pitch curves and asign them to Flight modes (IDLE1, 2, 3)
On input screen set a switch to Aux1
On output screen assign Aux1 to channel 5 (Gyro Gain)
Now you can use servo end points to adjust Gyro Gain and switch betwee HH (0 to 100) and RM (-0 to -100)

Adelaxie, 2018-01-05:
At present, the gyro value is switched by the flight mode and we will take your suggestion （switch control mode ）into consideration.Thanks