FrSky-OS/HORUS-X10-X10S-X12S-bug-report #167: Flap setup, no elivator compensation

Maverick440, 2018-01-14:
Ive come to set up my Carbon Cub which will have three stages of flap, this is easy enough to set up in FrOS however there is no ability to also add in some elivator compensation so that when you frop the flaps the elevator also moves to keep the plane in balance.
This feature is avalable on my Spektrum DX9.

Basicly its the same as the air brake feature where you can add elevator to compensate.

choochoo2, 2018-02-06:
Apparently I'm not understanding the problem.

I don't currently have a model programmed with flaps so I went into an existing model and assigned a pot and an output channel for FLP, OK flaps work on the monitor.  Then I went to page 3 of the airplane model settings to CMBFLP->ELE and turned it active.  OK now there is elevator compensation with flaps.  Isn't this all you are asking for?

Woita2095, 2018-02-06:
Well, it works as Choohoo2 said. 
But only if the flaps are controlled via potentiometer.

If you control flaps via 3-way switch, the flaps go too quickly. You can slow-down them via "SPEED" menu, but elevator goes quickly in this case - unusable.

So, what about the speed of the flaps vs. speed of elevator? See #174 ...

choochoo2, 2018-02-06:
The speed is a different issue, #174, and should be addressed.  This issue, however, only mentions the "ability to also add in some elivator compensation", which already exists.