FrSky-OS/HORUS-X10-X10S-X12S-bug-report #169: SYS/STICK DIRECTION selection bug

ssray0, 2018-01-15:
If you scroll past the first ten items and then select a “stick” there will be extra NORM and REV positions added.  So going to Extpot2, selecting it and scrolling right will go NORM to REV to NORM to REV. Going back up to J1 now will add another NORM and REV as the screen items will move down one position to get back to J1.  So there are now 6 NORMand REV you can move between.  Going all the way to the bottom (SWEXTR) and then back up to J1 allows you to scroll a lot to the right before it stops changing.
   This doesn’t appear to affect the actual function, as far as I could tell, and you normally just move a click forward or back so probably not an issue.  Really strange bug however!

Scott

Adelaxie, 2018-01-30:
Yes, it is.
only one parameter is scrolled when new page appear. Will try to make it better.
Thanks for feedback.

Adelaxie, 2018-01-30:
Fixed and please pay attention on v1.5.03 release

closed 2018-01-30 by Adelaxie