FrSky-OS/HORUS-X10-X10S-X12S-bug-report #175: electric brake

robertobuti, 2018-02-02:
Hello,
i have a suggestion to improve the system, provide some functions to manage electric brake, normally we use a slider to brake the plane  on dedicated channel, with proportional intensity. I like set an alarm if you forget the brake inserted for long time. (can't be ON too much, dangerous for temperature brake and battery), more a track player on slider treshold will be useful to advise you have brake inserted.

thank

Woita2095, 2018-02-02:
Well,
but this is only one separate thing.
Maybe would be better to have a timer, which could be started by a switch (of course, logical switch too). And then use the value of this timer  in functions, especially in the "Special functions"). This way would be able to playback track with warnings etc.

MIKE-DAILEY, 2018-02-02:
With the current X10S/X10 firmware that can be done with the delay function in the logic switches and in the next release for the X12S.

Woita2095, 2018-02-02:
So, this suggestion #175 looks to be solved then :-)

MIKE-DAILEY, 2018-02-02:
I would think so.  Just program a logic switch to monitor the brake channel with delay.  The logic switch becomes true with the brake on and the delay time true.  Add the logic switch to the special functions that plays a brake warring wav file.  Available now with the X10S.X10 and soon with the X12S,

Woita2095, 2018-02-02:
There is FW 1.5.03L for X12S, where it is done allready.

MIKE-DAILEY, 2018-02-02:
Yes, but that is beta and not released to the FrSky website yet.  Should be soon though.

Woita2095, 2018-02-02:
The logical switches seems to be very strong tool.
Certainly would be great if there was more LS's. 
8 of Logical Switches is too little. 32 would be better, 64 would be great!!!

Woita2095, 2018-02-02:
Agree. I just wanted to write:

"Maybe it worths to create a new issue, to increase number of the Logical switches?"

So I am going to write it as a new idea.

Adelaxie, 2018-05-09:
if anything else not resolved please reopen the issue. @robertobuti

closed 2018-05-09 by Adelaxie