FrSky-OS/HORUS-X10-X10S-X12S-bug-report #176: Increase to number of Logical switches

Woita2095, 2018-02-02:
Please increse number of Logical switches.
LS seems to be very strong tool, with their new functionality (delay, etc).
8 of LS is too little.
It would worth to increse number of Logical switches to 32 or (better) 64.

janekx, 2018-02-02:
Yes I think so.
If we wanted to program more complec behave we find that LSW are not much we need

MIKE-DAILEY, 2018-02-02:
I agree more logic switches would be great.

ssray0, 2018-02-17:
Yes, this would give more capability!

Lombardoa, 2019-05-12:
I agree, and I would be very happy to see two more pages of LSW in the next update.

Maverick440, 2019-08-11:
Yes we need more logic switches, I've already used all 8 for telemetery information and am now limited to what I can do with my X10S.
At least 16 would be good  with 32 being even better