FrSky-OS/HORUS-X10-X10S-X12S-bug-report #177: Manual Edits

choochoo2, 2018-02-05:
Hopefully this is the correct place to post this.

I'm new to FrSky products and not an expert in much of anything but I'm fairly good at catching typos and other errors in manuals.  The X10 manual is pretty good actually but could be better, so I've gone through the X10 manual and done a markup (attached).

Many of the suggestions are just correcting what appear to be typos or awkward wording.  There are also suggestions for areas that are unclear to a new user and need more/better explanation, and a few spots that seem to be mistakes.

I hope this will be helpful.

Rick

PS:  Unfortunately I am unable to attach the marked-up pdf file, all I get is "something went really wrong, and we can't process that file. Try again."  Is there a limit to file size or something else that might be causing this, or an alternate place or method for submission?

Adelaxie, 2018-02-05:
Please try to mail us : libo.xie@frsky-rc.com Adela Xie.
Thanks

Adelaxie, 2018-02-06:
Received and thanks .
will response to you by mail.

closed 2018-02-06 by Adelaxie