FrSky-OS/HORUS-X10-X10S-X12S-bug-report #182: 1.5.03 Wrong power options for the R9M module

OlivierC-FR, 2018-02-13:
When you have the radio flashed with the FCC firmware, and you flash the R9M with the LBT/euro firmware, it displays the powers option for the FCC (10 100 500 1000mW), when it should only list the two LBT 25 and 500mw.

The code is assuming the R9M module is running the same region as the X12S, it should not assume, it should check.

https://www.rcgroups.com/forums/showpost.php?p=39165303&postcount=14263

Adelaxie, 2018-05-09:
Please check it on latest firmware version,thanks for feedback.

MIKE-DAILEY, 2018-05-09:
@Adelaxie I think he is looking for a way to have his X12S flashed with NEU for the internal IXJT and run the R9M with EU.  There would need to be a check box on the RF SYSTEM screen for the external module section to select EU or NEU so the R9M power settings do not follow the main system firmware.

MIKE-DAILEY, 2018-05-09:
Need to verify but apparently 2.4g NEU is legal in some countries but 900mhz is not and it needs to be EU.

mariuszlabedzki, 2018-08-29:
I have the same issue :( is there any chance to fix this ?

MIKE-DAILEY, 2018-08-29:
Is this selection what you wanted?  https://www.rcgroups.com/forums/showpost.php?p=40127696&postcount=2147

mariuszlabedzki, 2018-08-29:
yes ! Thank you. I saw open version but I was not sure what  changes are compared to normal one - is there any list ?