FrSky-OS/HORUS-X10-X10S-X12S-bug-report #183: Possibility to copy flight modes in FrOS software 1.2.03 and 1.2.03.

liukku56, 2018-02-15:
I want to be able to copy flight modes so I do not have to redo all settings again for every flight mode.

Hardware: X10 EU14
SoftWare: 1.2.03
PhysID: 23
---

janekx, 2018-02-15:
You mean copy FM from model to model? Why not copy model and change other items.

liukku56, 2018-02-15:
Hello janekx!
No, I mean to be able to copy flight modes for the present model.
---

andymeade, 2018-02-15:
Similar to my #96 - every flight mode needs expo rates etc setting up very carefully.  Ideally needs a copy function!

liukku56, 2018-02-15:
Thank you andymeade, I mean exactly as you point out.
---

janekx, 2018-02-19:
duplicate to #96

closed 2018-02-19 by janekx

stanley1211, 2018-02-23:
This is urgent needed. I found it very tedious to program without this copy function, and it's easy to make misstake during programming