FrSky-OS/HORUS-X10-X10S-X12S-bug-report #184: mode select not works in FrOS for X12S ver. 1.5.03

lucamini, 2018-02-15:
I updated my Horus X12S with the 1.5.03 FW and when I create a new model, I need configure it as mode 1 as stick configuration. When I load the model it is not right, mode remains mode 2.
I tried both firmwares contained in the zip files (EU release), mode1 and mode2 with the same results.
In short, mode cannot be changed, graphically it is shown correctly, butit does not works.

Luca Mini

janekx, 2018-02-15:
Did you change the config files?

lucamini, 2018-02-15:
I can describe exactly what I've done:
- my H12S is new (arrived 3 days ago)
- downloaded and extracted the zip file
- put in a new uSD a folder FIRMWARE and the file X12S_mode1_EU_1503_frtx.bin
- installed as the documentation explain

then, I tried to install X12S_mode2_EU_1503_frtx.bin with the same issue.
In the zip file there's no config files.

PS:
However there are some incongruences between documentation for flashing, manual, names of files and unclear contents of the zip file (i. e. in the zip file there are two folder named "1.5.03" and "1.5.03L" without nothing to explain what they are and how to use it) that may confuse users.

janekx, 2018-02-19:
L FW means are other language enabled FW, if you go with English you not care about L ones. 
In fact there is each model creation question about mode the mode1 and 2 is little unsense now. 
Config files are here in the zip file rewrite yours in Flash 
[CONFIG.zip](https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/files/1737949/CONFIG.zip)

Adelaxie, 2018-04-24:
@lucamini  If the config file resolved your problem?

Adelaxie, 2018-04-25:
thanks for your response. I will close the issue ,then.

closed 2018-04-25 by Adelaxie