FrSky-OS/HORUS-X10-X10S-X12S-bug-report #185: Telemetry GPS Speed

mdrjr, 2018-02-21:
I have a plane with X8R and FrSky GPS V2 on SmartPort.

Everything works just fine I can see the speed on my X12S screen, I decided to start logging my runs so I could plot it later however I don't understand the "spd-g" (speed gps) values.. I know for sure that on a flight I did 168km/h (radio logged that on screen) however I have a peak of 65000 on spd-g telemery log.
In what unit is that 65000?

Q: what's the actual unit that it output to the CSV file?

janekx, 2018-02-23:
Do you have onle some peaks or continues you got bad data?

mdrjr, 2018-02-24:
Here's the full CSV of a flight..

https://pastebin.com/bLx4WRbd

mdrjr, 2018-02-24:
A good idea is to add on the logs and telemetry screen on the radio how many sats we have locked

janekx, 2018-02-25:
thnaks will look on lof shortly, first looks strange as datas are too big and jumping

janekx, 2018-02-25:
Do you have connected only GPS or other sensors ?

thock1960, 2019-04-09:
In trying to determine the units for the GPS speed in the log file, I ended up setting up the X10S and a GPS receiver unit only connect to a receiver to compare the speed on the radio with the speedometer on the car.  I was on a long highway stretch, set the car cruise to 40 MPH, started logging data and also noted the X10S showed 40 MPH, very good agreement between the two which gave me confidence in the test. The log file did not show 40 MPH or km/hr, knots, m/s or any other possible unit I could think it might be. However, the correction I came up with was to multiply the results in the log file by x0.74513 to convert to MPH. which seems to work well.
Either there is a bug or the documentation needs to list what are the units for the GPS velocity in the log file.