FrSky-OS/HORUS-X10-X10S-X12S-bug-report #186: Elevator not synchronized for butterfly deployed when offset of center in glider mode

stanley1211, 2018-02-23:
For glider, we set crow braking (butterfly) to slow down the plane. In the Butterfly manual we just mix the flap, aileron with some down elevator, fine it can be done. However, the assigned BFLY (throttle stick) is deployed only when it is half way down, i.e., not using its full travel. OK, fine I then try to modify the RATES/EXPO for BFLY input with offset of center location on top to solve this issue. Well, it works in a right way for AIL and FLP when I pull the stick from the top all the way down, however, the elevator is not  moving until the stick is half way down!  That means the elevator is not synchronized with the butterfly on the first top half of the stick movement (a bug obviously). I didn't notice until I flew my model trying to land with crow braking, when I start pulling the butterfly stick down, the plane shot up because of no elevator compensation, only when the stickis half way down then the elevator is kicked in, but it is too late, I almost crashed my model as a result!  Please check and correct the bug, thank you!

liukku56, 2018-02-24:
I agree with **stanley1211** but I am a lucky guy who don't want the left stick (J3) to kick in for crow before middle position, so I am OK. I use full travel for the flaps when crow braking. My FrOS is also 
the latest version 1.2.03.
/Ville

[image]

stanley1211, 2018-03-20:
Yes, I do tempolary use the EX mixer, 6 channels of mixing, to solve the problem, but it is very inconvenient to program and if slight change is needed to make. Anyway, the problem I mentioned is regarding I use OFFSET in RATES/EXPO for BFLY stick. When I offset the curve to start the stick at top while output is at 50%, all control surfaces move accordingly except the elevators which still start at the middle of stick position, that must be a bug, please check! I want to upload a picture to show you the offset I did, but don't how to upload it!

Woita2095, 2018-03-20:
Just "Drag and Drop" your picture here :-)

liukku56, 2018-03-20:
I understand **stanley1211**,  but I don't know if this issue is a bug, it seems that FrSky don't think so because they have not categorized this as a bug low or bug critical, yet, they have not put any label on this issue.
I don't like the default setup we get when we create a GLIDER. FrSky does not tell us where we should start when we are going to program eg F5J model, which may be important for a beginner, I can handle myself because I've been in the game for so long. In the manual they say that there will be a more detailed instruction in the "setup guide". Which "setup guide"?
Now I will stop complaining. I could describe how to configure an F5J glider plane, from A-Z, and maybe I do, but not here.

/Ville

stanley1211, 2018-03-21:
I set the offset and weight in RATES/EXPO to bias the BFLY curve so that all control surfaces should start to move at the top position of the stick, this is because the default butterfly activation is at 50%. Anyway, all control surfaces move accordingly except the elevators which still activate at the middle of the stick.  If it is not a bug, then the logic that allows offset and weight to control input is meaningless. Please correct this bug! 

My butterfly setting is shown in below, -15% for both elevator. 

[image]

[image]

[image]

liukku56, 2018-03-21:
Hello **Stanley1211**!
I can agree with you that elevator control surfaces should follow BFLY (throttle stick) when you pull the throttle stick. But if it's a bug that they do not, I'll let someone else decide. I believe that whatever method you use with today's X10 software when using BUTTERFLY, you must also use an EX mixer to get the elevator control surfaces to follow the BFLY (throttle stick).
I see that you get enough throws on the flaps using only half of the possible servo travel. It has never happened in any of my F3J / F5J machines.

I mainly use Multiplex radio systems and I have borrowed some text from one of their manual, explaining the work of **servo calibration** of flap servos which is a necessary step for almost anyone who is programming a F3J / F5J model. I think they explains it in a good way. 
All servons in a model must be calibrated but servo calibration at all is often a forgotten chapter. With X10 (FrOS) I always start servo calibration of the flap servos in the butterfly menu for my gliders.

**Special features when calibrating the servos for FLAPS, (and some times even for the AILERONS)**

Models with a four-flap wing generally employ what is known as the Butterfly or Crow system as a landing aid: ailerons max. up, camber-changing flaps max. down. For this function the flap servos, in particular, have a highly asymmetrical working range:
The maximum control surface travel in the updirection (approx. 20°) is required for aileron control.
For landing the flaps are required to deflect down as far as possible in order to achieve maximum
braking effect (if possible > 60°). This means that the servo travel must be greatly
reduced in the up-direction.

1. Install the servo output levers for the flap and aileron servos at right-angles to the pushrods.

2. Determine the centre of the working range of the control surface:
For example: the working range of the control surface (e.g. camber-changing flap) is from
+20° ... -60°, starting from the neutral position (flap in line with airfoil)
* The centre of the flap’s working range is therefore -20°.
Now adjust the flap pushrod so that the flap is at the -20° point when the servo is at centre.

3. Now you go to the BUTTERFLY menu and adjust the throws for the flaps servos so that you get
the control surfaces to neutral/normal when left stick is at the max up position, and also adjust the settings so that you get max throw, -60 degrees or more when left stick is at max down position.

[image]

[image]
---

stanley1211, 2018-03-23:
Thanks, liukku56 for your information.

I fully understand the operation and setting of butterfly for crow braking as I participate in many F3F competitions which crow braking is a very important part. The problem X10S has is related to the synchronization of elevator movement with the rest of other surfaces. If I just use the default butterfly setting which activates at mid stick position, that's fine, everything move accordingly. But if I offset the activation position to the top of the stick by using the OFFSET and WEIGHT in RATES/EXPO of BLFY input, other control surfaces start moving at the top but elevators still start at the middle, that's is the problem! I never think this was designed like that, simply wrong logic!

liukku56, 2018-03-23:
**Hello stanley1211!**
I think we are totally in agreement with the problem. In the MPX system, there is no "offset" menu like in FrSky X10, offset functions is (mostly) done in the configurable mixes.
The reason I describe the offset function and servo calibration so carefully is because there are certainly readers who are beginners and who can benefit from the information.

/Ville

PistiPisti, 2018-06-15:
Hello Stanley1222,

I have almost the same problem. Flaps are moving immediately, but my ailerons not moving until the left stick is half way down. (Flaps and Elevator moves after left stick is half way down, but it wouldn't be a problem for me. The problem is that my ailerons not moving until left stick is half way down.)
Could you tell me how can I modify the RATES/EXPO for BFLY input with offset of center location on top to solve this issue? I do not know how to do it.

Thanks in advance,

Pisti