FrSky-OS/HORUS-X10-X10S-X12S-bug-report #187: Throttle hold function has changed for petrol and nitro models !!!!!

GIB2A, 2018-03-04:
Hello
I was part of the beta testers in Europe and we have raised several times already the problem of the function Throttle hold which is not suitable for use in flight on our models
The Throttle hold function is locking the throttle and it is therefore impossible to have a slowdown slightly higher in flight (example futaba t12mz)
this function is really useful when flying on gasoline or nitro machines
is it possible to be able to unlock this function?

MIKE-DAILEY, 2018-03-04:
From your description it sounds like you do not want "throttle hold", but "idle up".   "Idle up" would have a screen that looks like the "throttle hold" screen, but would set the “idle up” low point of the throttle so that even if the throttle stick it moved to -100 the channel output would only reduce to the “idle up” set point. This function keeps the engine speed from going below the fast idle or “idle up” position during manures and reduces the possibility of the engine from quitting.  With “idle up” turned on the throttle works normally between the “idle up” position and full throttle.

GIB2A, 2018-03-04:
Hello
thank you for your interpretation which is more jsute than my description !!!! my english is not very good sorry
indeed the function TRH HOLD of frsky does not allow an IDLE DOWN similar to the program futaba for example
it would be wise to add to the function the posibiliter from TRH HOLD to IDLE DOWN
or create this function
cordially

MIKE-DAILEY, 2018-03-04:
I'm not sure I understand, there is "throttle cut" that works just like Futaba.  "Throttle cut" overrides the idle trim lever on the throttle and cuts the engine.  Is that what you want?

GIB2A, 2018-03-04:
The Throttle cut function is not a problem
The Throttle hold function blocks idle at a fixed accelerated point!
it lacks an IDLE DOWN function that allows a second idle engine and does not lock the throttle
(The function idle down is a second gas trim that is activated by an inter defined in the function idle down see manual futaba t12mz t14mz ect ....)
exmple:
No function: Engine idling 1 rpm for taxing and landing
Idle down: Engine idling 2 engine regmie (500 more tr/min) for the flight
this function is not present in FrOS and it is a shame considering the development of the new receivers pro and accesoir for gasoline engine.
We have already reported this to the importer europe but I think the information has not recovered.

MIKE-DAILEY, 2018-03-04:
I think I understand.  What you call "Idle down": Engine idling 2 engine regime (500 more tr/min) for the flight" "idle down" I call "idle up". 

"Idle up" makes engine idle faster more RPM like I think you want.

slotus7, 2018-03-05:
387/5000
Hello
I am also a former beta tester Europe Belgium.

Gib2a is right, it's a good idea. the accelerated gas function is required in flight according to certain phases of flight to keep a higher engine speed.
Nevertheless, the current safety function to limit the engine speed must be maintained. Maybe two function, ground and in flight.

Regards

Patrick

slotus7, 2018-03-05:
Hello Mike-Dailey
Ok , that's it, more rpm with a function switch.

MIKE-DAILEY, 2018-03-05:
@Adelaxie It would need to be a new function called "idle up".  The screen would look like the throttle hold screen but the difference would be that when the POS was set the throttle could move from up from that point to full but not lower.

GIB2A, 2018-03-05:
Hello Mike-Dailey
Thank you for your answer that's fine with a switch or flight mode!

It also lacks the multi-engine function in the FrOS soft to manage more idle engine. this function is very useful for gasoline or glow models
very cordially
Etienne