FrSky-OS/HORUS-X10-X10S-X12S-bug-report #188: EMERGENCY DANGER USE ENGINE IGNITION RCEXEL

GIB2A, 2018-03-10:
EMERGENCY DANGER USE ENGINE IGNITION RCEXEL

Hello
I come this morning to lose a plane equipped with an X8R with a rcexel ignition
this problem has appeared on several receivers and models at the club and obliges us to stop the flights of gasoline models equipped with FRSky
Here is the problem :
The ignition hides the reseption and blocks the recptor which becomes inactive, the phenomenon this product has to bear different decision according to the ignitions
mine about 150 meters (engine glow 4t converted) that of another college month of 20 meters (dle 50) and finally the one who is riding on a dle 30 c is when starting engine route after 30 seconds
this is a radio propleme because we have all swap our emitter and our receiver for test (horus x12s and x8r x6r rx8x) and the phenomenon is the same
it is desirable that FRsky communicate quickly about the problem. Who is already identified on the forums !!!!!!!
I can put on line a video of the demonstration of the problem !!!!!! by proving that this is a frsky (I'm going to the demonstration with futaba or spectrum to the choice !!!!!!)
cordially

GIB2A
BETA TESTER FRENCH CORSICA #

MIKE-DAILEY, 2018-03-10:
Are you using the X8R 170210 firmware that was released March 07, 2017 that enhances anti-interference capability when operating with defective ignition systems or installations emitting high levels of RF noise?   https://www.frsky-rc.com/x8r/

FrSky has announced a new receiver designed to operate in noisy RF environments.

GIB2A, 2018-03-10:
Hi MIKE-DAILEY
I use the latest version of software for both radio and receivers
this ignition is also new
We encounter the same problem that begins to go up in France, here is the forum that deals with the subject
http://frskytaranis.forumactif.org/t4882p50-blindage-x8r
http://frskytaranis.forumactif.org/t6434-nouveaux-rx-pour-les-moteurs-thermiques
A QUESTION ABOUT THE FORUM
does not the problem come just a crash of the receiver processor .... I do not know the clock frequency of this processor, but it seems to be sensitive to the harmonics of a spark ignition?
cordially
thank you for your implication to understand our difficulties !!!!!

slotus7, 2018-03-10:
Hi Mike-DAiley , Hi GIB2A
Does using Banggood's 90 Degree Antenna Holder Have Been Problematic?
Maybe the antennas are too close and in the ignition radiation in some flight configuration.
Regards
Patrick

robertobuti, 2018-03-11:
I see the X8Rplus on FRSKY web site with ingition interference immunity capability, so somethign has been done.......
Plans to apply same adjustment on S8R type / family ?

Waiting for receiver with improvd immunity i suggest apply noise filter on ignition power supply like these :  [LC noise filter](https://www.ebay.it/itm/LC-FILTER-16V-2A-FPV-Power-Filter-Noise-Eliminator-for-RTF-RC-Quadcopter/183076898812?_trkparms=aid%3D222007%26algo%3DSIC.MBE%26ao%3D2%26asc%3D20140620074313%26meid%3D474c91a190a348a19c9ab82ac5df5fe0%26pid%3D100011%26rk%3D2%26rkt%3D2%26sd%3D322089730483%26itm%3D183076898812&_trksid=p2047675.c100011.m1850)

I have checked pratically on my plane and tested with scope, help on protection.
Best is use separate power supply / battery for ignition if you like this solution. (i don't like it too much)

GIB2A, 2018-03-11:
Hello robertobuti
Thanks for the links I will do a test with the noise reducer.
all my planes are fitted with separate batteries
cordially

MIKE-DAILEY, 2018-03-12:
The new RX8R Pro is now available https://www.horusrc.com/en/frsky-rx8r-pro-receiver.html

Adelaxie, 2018-05-09:
@GIB2A  Anything else  not resolved please reopen the issue.

closed 2018-05-09 by Adelaxie

ululi1970, 2019-04-23:
I experienced the lockout problem with a X8R flashed with the latest firmware, on a rcexel ignition on a separate battery (with opto kill switch). Had to cycle RX power off and back on to regain control. So it appears that the firmware fix does not fix the problem after all