FrSky-OS/HORUS-X10-X10S-X12S-bug-report #189: logical OP in LSW

robertobuti, 2018-03-12:
Hello,
i like logical op as "and / nand / or / nor" to be used in LSW menu, more then the LSW result need to assume different value, not only true or false.
Pratically for example i need to condition an output (like 3 position switch) from one master source, like Speed, altitude or any other. 
Target is have a command (es. 3 pos switch) enabled only in special conditions. (es. ground , not in flight) to prevent dangerous accidental activations.
My be my lack but i don't find this possibility on actual sw revision.

bye

choochoo2, 2018-04-02:
Since no one else has commented on this, let me try.

If I understand your request correctly, the capability already exists.  For instance, suppose you want to make sure you don't accidentally raise the landing gear while the plane is on the ground.  You can set logic switch 1 (LSW1) so it is true if alti-v is greater than 20M.  Then set LSW2 so it is true with LSW1 AND your gear-up switch choice, perhaps switch-F down.  Then use LSW2 as the input for your gear channel.

With this setup, the gear switch only operates when the altitude is greater than 20M.  On the ground, the gear can't be raised.  Also, if you forget to lower the gear when landing, it will automatically lower when the plane descends below 20M.

I would like to post a screen shot but attaching photos doesn't work for me on this site.

Rick

strgaltdel, 2018-04-02:
Ich kehre zurück am 03.04.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] logical OP in LSW (#189)" gesendet am 02.04.2018 04:51:14.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.