FrSky-OS/HORUS-X10-X10S-X12S-bug-report #190: Control mapping bug

NevinHK, 2018-03-19:
Just updated software today but problems still exist.

Input map Elevator = J3
Output map CH2 = Elevator
Mode 1

Elevator is plugged into CH2 on receiver.

When I move the elevator stick nothing happens.

When I move the aileron stick sideways, the elevator moves!

If I change input to Aileron =J3, I can then control elevator (but in reverse).

Also, lost model alarm was working fine, but now alarm is on continuously. This is the same for all models with aileron/elevator setup. Doesn’t seem to effect flying wings that I’ve got.

I’ve tried two different new receivers. Any help appreciated.

MIKE-DAILEY, 2018-03-19:
Can you please provide more information about the FrOS firmware level your system was on when it was working OK and the versions that you have updated the system with?

NevinHK, 2018-03-20:
I only did the upgrade yesterday for the first time. When I first brought the X10s it seemed to work fine. Then about a month ago (with no upgrade) I noticed that my lost model alarms stopped working. So it don’t think it’s an upgrade issue.

Last week I needed to replace a receiver, and I noticed the problem described, that none of the controls were mapping correctly. I tried another new receiver, with the same results. Yesterday I performed the upgrade to latest release and tried both receivers, and also a third receiver (also new) with the same results.

However, I then tried another glider with a G-RX8 receiver, and it works fine except the lost model alarm cannot be controlled. I’ve then checked to see that when I switch between models the mapping remains the same (and it is).

So I’m not sure if it’s something I’m doing incorrectly. However, previously, all my models were fine and I hadn’t changed anything in between. So it’s a bit baffling.

Thanks.

NevinHK, 2018-03-20:
> OK so it is a X10S system. Try creating a new model and then check if the default input and output mapping is working OK before you change anything.

Yes I tried that several times. The default mapping looks to be correct each time. But the problem still exists.

What I will do is try a different model. This is because one of my existing models seems unaffected.

Thanks.

219 Seabee Lane
Discovery Bay
Hong Kong.

Tel: +852 9327-8442

MIKE-DAILEY, 2018-03-20:
When I say create a new model that is not the same is copy a model.  We want to find out if you create NEW model that the mapping is correct.

NevinHK, 2018-03-20:
Apologies. I had actually done that too, but just to be sure I’ve created another new model. Will attach a screenshot for you.

NevinHK, 2018-03-20:
[image]
[image]

MIKE-DAILEY, 2018-03-20:
Those screens look OK and you selected Mode 1.   Does it work OK now with a new model in Mode 1?

NevinHK, 2018-03-20:
No, unfortunately the exact same results. Yet when I change back to the G-RX8 it seems to work fine.

219 Seabee Lane
Discovery Bay
Hong Kong.

Tel: +852 9327-8442

MIKE-DAILEY, 2018-03-20:
Look at the Channel Monitor.  Are the outputs correct there with the new model?

NevinHK, 2018-03-20:
Yes, on the channel monitor everything is correct. For example ch2 is elevators which matches the output map.
But when a servo is plugged into ch2 it is the aileron stick that moves it, even though ch2 output is set to elevators.

janekx, 2018-03-20:
We still talk about GRX8 that you see ele on CH2 and you move ele and CH2 are moving on monitor, but on RX the ch2 is not move ? What chanel move? what imput move with ch2?

NevinHK, 2018-03-20:
When I move the elevator stick, no servos move at all. When I move the aileron stick, the elevator servo (CH2) moves. Yet on the monitor it is not moving.

219 Seabee Lane
Discovery Bay
Hong Kong.

Tel: +852 9327-8442

janekx, 2018-03-20:
You sad on monitor is all OK 
so on 
AIL is maped to Ch 1
Ele is maped to Ch2 

monitor AIL Ch1 and Ele Ch2 move respected channels 
RX AIL move with ch2 and ele not move any of channel?

MIKE-DAILEY, 2018-03-20:
The channel monitor shows what the X10S is outputting on the channels and what would be sent on the RF link.   If the stick inputs you make are correct on the channel monitor then it is most likely not a problem with the X10S.   The only other control that can have impact on the channels sent on the RF link is how you have the reviver bound to the X10S and the channel selection you make there.   The Normal setting would be CH RANGE 01-16.

NevinHK, 2018-03-21:
Thanks! You are right, I had incorrectly selected the wrong channel range for the new model. Once I made the correct select all the mapping at the receiver matches the transmitter. Many thanks for helping me with this.

I still have an issue with the lost model alarm - it works on some receivers but not others. The transmitter setup for it seems to be identical - using the Landing gear mapped to channel 6 for instance, switch B to activate from a range of 125 (up) to -125(down). Alarm will not deactivate on the G-XR8. Monitor shows the switch changing values correctly on the correct channel.

janekx, 2018-03-28:
I think that question of GRX8 and alarm is not related to radio FW, so please close the issue.

strgaltdel, 2018-03-28:
Ich kehre zurück am 03.04.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] Control mapping bug (#190)" gesendet am 28.03.2018
13:33:01.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

closed 2018-04-24 by Adelaxie