FrSky-OS/HORUS-X10-X10S-X12S-bug-report #191: New input - ON

MIKE-DAILEY, 2018-03-28:
Add a new input that is named ON that is active when the system is powered on that can be used with logic switch functions, etc.

strgaltdel, 2018-03-28:
Ich kehre zurück am 03.04.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht
"[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] New input - ON (#191)" gesendet am 28.03.2018 03:27:38.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

MIKE-DAILEY, 2018-03-28:
Automated out of office response from strgaltde deleted.

strgaltdel, 2018-03-28:
Ich kehre zurück am 03.04.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] New input - ON (#191)" gesendet am 28.03.2018 11:58:28.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

Adelaxie, 2018-04-24:
@MIKE-DAILEY  How to use? please describe it.
ON that is active when the system is powered on but system off there is no running functions. so ON will be always 1 when system running,right?

MIKE-DAILEY, 2018-04-24:
@Adelaxie Correct "ON" is always on and can be use to do things like play the name of the aircraft selected when the system starts or any other logic function you want to do when the system starts.  it is a feature in OpenTX.