FrSky-OS/HORUS-X10-X10S-X12S-bug-report #192: backup

Salvatico, 2018-03-30:
It would be very useful to insert, in model select menu, the possibility to save (backup) the model on the SD card. Of course you also need the restore button.
(As with Taranis)
Grazie

Woita2095, 2018-03-30:
But you already have your models on the SD card, they are in the SYSTEM folder. 
And you can make copy of your model at SYSTEM - SELCET MODEL menu, if you want to do experiment with some of your model.

strgaltdel, 2018-03-30:
Ich kehre zurück am 03.04.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht
"[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] backup (#192)" gesendet am 30.03.2018 17:18:56.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

strgaltdel, 2018-03-30:
Ich kehre zurück am 03.04.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] backup (#192)" gesendet am 30.03.2018 17:56:01.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

MIKE-DAILEY, 2018-03-30:
The normal place for the model files in the flash memory drive,

Woita2095, 2018-03-30:
Yes, Mike-Dailey, you are right, I apologize. I always mistaking what drive is Flash and what is SD.
But anyway, the models are in the Flash memory, thus user can make a backup to his PC. As to me, there is not reason to do a copy of models to SD card.

strgaltdel, 2018-03-30:
Ich kehre zurück am 03.04.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] backup (#192)" gesendet am 30.03.2018 18:34:15.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

strgaltdel, 2018-03-30:
Ich kehre zurück am 03.04.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] backup (#192)" gesendet am 30.03.2018 18:43:14.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

Salvatico, 2018-04-01:
Dear Woita
When you want to make changes to a model, it's a good idea to make a save first.
Of course it is now possible to make a copy of the model, rename it, then resume it with the changed name and then modify it. But this operation is inconvenient and also takes up memory on the Flash. Furthermore, in the case of an internal memory problem (for example during an update), you may risk losing the model data. I remind you that a backup similar to the Taranis that indicates the date of the saving, allows you to create a model history. The solution to connect the PC is certainly viable, but even this is much more cumbersome than a very simple beckup / restore.
Sorry for my bad English
Best wishes
Paolo

Woita2095, 2018-04-01:
Well, Paolo, I agree with your argumentation, you are right. Maybe it is not bad idea to have this functionality. Certainly it is not necessary, but useful.

strgaltdel, 2018-04-01:
Ich kehre zurück am 03.04.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] backup (#192)" gesendet am 01.04.2018 09:47:33.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

strgaltdel, 2018-04-01:
Ich kehre zurück am 03.04.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] backup (#192)" gesendet am 01.04.2018 10:01:19.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.