FrSky-OS/HORUS-X10-X10S-X12S-bug-report #193: Logging drops data frames

choochoo2, 2018-04-05:
The .csv  log files created during flight are missing data entries.  The log from yesterday's flying session contained data from three flights on two aircraft.  For one aircraft the log was missing 8% and 7% of log entries on each of two flights respectively.  The log for the other aircraft was missing 17% of log entries.  On both aircraft the logging was set to half-second intervals or twice per second.  Transmitter is an X10s.

The missing entries are apparently at random but often in groups.  In some cases a couple of seconds of data are missing entirely.  Such missing data distorts graphs and can greatly undermine the value of logging in evaluating problems if the missing data occurs at an inopportune time.

choochoo2, 2018-04-05:
> Need more info about your model, ideally your OTX file and details about any lua scripts you're using.
>
> But SD bandwidth is low, shared between a lot of stuff, and logging obviously is low priority.

It’s FrOS, not OTX.  I thought the github error reports were only FrOS.  

 

Others in the RCG forum mentioned they saw similar problems in OTX but my observation was FrOS.

https://www.rcgroups.com/forums/showpost.php?p=39457693 <https://www.rcgroups.com/forums/showpost.php?p=39457693&postcount=7770> &postcount=7770

 

Rick

choochoo2, 2018-04-24:
Not sure what additional data is needed.  There is a sample log file attached to the original post.  With the logging set for half second intervals there should be two data points logged for each second of elapsed time but missing data points are scattered throughout that file.  

For example, the log starts out OK on line 2 with two data points logged each second.  On line 12, however, only one data point is logged at 10:34:40, and no data points were logged at all at 10:34:41.  This erratic dropping of data points continues through three flights on two aircraft.  

On the SkySurfer flight, 17% of the expected log entries are missing.  That's an awfully high drop rate.  It interferes with troubleshooting and with accurate graphing.

What more do you need?

Adelaxie, 2018-04-25:
We will do tests. thanks.