FrSky-OS/HORUS-X10-X10S-X12S-bug-report #195: MDL Model menu pg 3 not returning to pg1

Mordee, 2018-04-07:
I push the MDL key and arrive at the model menu, and page through to page 3 then on next push pages do not cycle back to 1, as they do in all other menus. It beeps but no page change, a long push takes me to page 2 (skips pg 1 ) then only way to go back to pg 1 is to push MDL key. Any ideas would be great, Thanks

New Radio with Software 1.2.03 Hardware X10 NEU14

choochoo2, 2018-04-07:
The long push is doing what it is supposed to do, back up a page.  You're right, however, that all the menus should cycle back to page one after the last page.  I've noticed this also and I think it might be missing on some other menus also.  It might also be convenient sometimes if a long press on page 1 cycled back to the last page, never tested this.

Mordee, 2018-04-07:
Thanks, Model Menu is the only one I have found affected.