FrSky-OS/HORUS-X10-X10S-X12S-bug-report #196: Improve request Too much backlight lcd to night flight.

flaviomartins8, 2018-04-21:
Hi, 
I fly a lot at night with my Led lights on my airplane.
The light on FrSky Horus 10Xs is too high in the minimum , either with no backlight ON.
Can it improve to less as possible or turn off completely the lights?

Tks

MIKE-DAILEY, 2018-04-21:
Hare you tried the sleep function that turns the display off completely?   Look at SYS/DISPLAY, select SLEEP and then set KEY for wakeup.

Adelaxie, 2018-05-09:
@flaviomartins8  If anything problem please reopen the issue.

closed 2018-05-09 by Adelaxie

flaviomartins8, 2018-05-09:
Hi, 
I tried several times and i do not found the sleep mode in taranis X10s with opentx-x10-2.2.1(cca 71808) eepr:218
Backlight:
Off
duration 0s
On and Off Brightness are in the min.
I do not found any special function to turn off completely the light 
we only can use min  to max light.
in the minimum, the light is too much for night flight.
Tks

flaviomartins8, 2018-05-09:
Can you open again the case for improve?

MIKE-DAILEY, 2018-05-10:
For OpenTX you would need to do that with the OpenTX developers.