FrSky-OS/HORUS-X10-X10S-X12S-bug-report #198: Spektrum DM9 module support with 1.2.04 Euro LBT firmware

Maverick440, 2018-04-27:
I use a Spektrum DM9 modue in the back of my Horus in order to fly some parkzone planes using DSM2 protocol.
Im using the EU LBT version of the Horus X10S and the module worked fine on the original firmare suplyed with the horrus back in December.
Ive now updated the fimware from the FrSky website to version 1.2.04 Euro LBT and my DM9 module will no longer work (No LED elumination showing no PPM signal)
I have checked my DM9 module by tying it in the back of a fellow club members Teranis and it works fine.
Interestingly I have tryed an OrangeRX module in the back of my Horus and that works so the update seams to only affect the DM9 module.
I have double checked that the RF is set to external module ON and the modulation is set to PPM-

Adelaxie, 2018-05-05:
You have used frOS with DM9 well? what was the version number?
We did not test any 3rd part module , we will check the difference between two versions if you can help to provide.

Maverick440, 2018-05-08:
Hi Adelaxie, i bought my Horus X10S V2 (internal charger) back in December last year and the DM9 module would work as i used it to setup some models that would not take an FRsky recever, unfortunatly i dont fly these that often so not sure when the change took place.
Ive tryed reloading the firmware to versions 1.2.03 and 1.2.02 but i cant get it to work on eter of these now and im sure it was shipped with 1.2.02