FrSky-OS/HORUS-X10-X10S-X12S-bug-report #200: Trainer / Pupil MZ18 and FrSkyHorus X12S - Channel Option "Replace" doesn't work

scvhorus12, 2018-05-11:
My configuration is MZ18 as pupil and FrSky X12S (FrSky OS) as master and connected via cable. Connection works if switch is defined on FrSky OS - that's true for channel options "OFF" and "ADD". But "channel option "REPLACE" doesn't work - even if no switch is defined.
If no witsch defined to activate master - no connection ! 
But I expect that REPLACE option should work if no switch is defined - that's the way I understood manual. 
Or make it work if switch is set.

scvhorus12, 2018-05-29:
My installed SW Version is 1.5.03L

strgaltdel, 2018-05-29:
Ich kehre zurück am 11.06.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] Trainer / Pupil MZ18 and FrSkyHorus X12S - Channel Option
"Replace" doesn't work (#200)" gesendet am 29.05.2018 11:58:47.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

Adelaxie, 2018-05-30:
@scvhorus12  how it work if you set a  switch ?

Adelaxie, 2018-05-30:
>But "channel option "REPLACE" doesn't work - even if no switch is defined.
If no witsch defined to activate master - no connection !
But I expect that REPLACE option should work if no switch is defined - that's the way I understood manual.
Or make it work if switch is set.

We design it as that only works when switch is set. If it works when there is switch set?
if there is no switch is set , the slave signal will not be used.

strgaltdel, 2018-05-30:
Ich kehre zurück am 11.06.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] Trainer / Pupil MZ18 and FrSkyHorus X12S - Channel Option
"Replace" doesn't work (#200)" gesendet am 30.05.2018 08:22:32.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

scvhorus12, 2018-05-30:
REPLACE option doesn't work if switch is set. ADD option works well in that case.
br siggi

strgaltdel, 2018-05-30:
Ich kehre zurück am 11.06.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] Trainer / Pupil MZ18 and FrSkyHorus X12S - Channel Option
"Replace" doesn't work (#200)" gesendet am 30.05.2018 09:42:07.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

Adelaxie, 2018-05-31:
@scvhorus12  We tried with 1.5.03L and it works.
Can you provide a link of video if its ok for you/

scvhorus12, 2018-05-31:
Hello I'm really wondering, and of course I'm interested to understand it in the same way as you did :-)  
You will find a video that shows my setup - here as download link, because it's more than 10MB. 
https://my.hidrive.com/lnk/f8gMwiaA

If there is any question let me know 
thanks in advance and br 
Siggi

Adelaxie, 2018-06-01:
@scvhorus12  have not downloaded it yet///////

scvhorus12, 2018-06-05:
Hello serviceteam, 
I would expect that option "Replace" will work like that:
* Control will be given 100% to master if  Control-Unit (Stick or switch)  at master has been moved. Just overtake control from pupil TX to master TX for that channel due using assigned control-unit (at master). 
This behavoir would have the option to control the flight object in a smart way - even pupil is doing absolutly strange things. Out of my view - Using "ADD" (with 80%) is fine as long as control direction is in a proper way. If it isn't  - for example pupil is given rudder left (but should be right) with 80% - master's move to right needs 80% to compensate wrong direction and only 20% will be available to change direction. That could be less to rescue aircraft.          
Current behavoir is like "ADD" and 100% and no option to interact from master.    
Hope it will help to have same understanding 
thanks and br siggi

scvhorus12, 2018-06-11:
Would you like to give me a feedback about your decision. And one question - is it possible to get it using logical switches?
thx in advance and br siggi

scvhorus12, 2018-06-11:
Sorry, won't to close it :-)

Adelaxie, 2018-06-15:
We have some discussions about your suggestion. will collect more comments on this before do something. thanks.

Woita2095, 2018-06-16:
Svchorus: My thought about taking control over via stick:

What would be the process to give the control back to the pupil, when the stick movement is used for taking over control? The master should have move the stick to the center and this way to return the control back to the pupil? It could be problem. 
When a switch is used for taking over control, it is O.K. (without problem))

Woita2095, 2018-06-17:
And how the master returns the control back to the pupil?  He just moves joysticks to the middle positions and in this moment the pupil can control the plane again?

scvhorus12, 2018-06-18:
It's little bit tricky . I would implement a control loop at every master stick marked as replace. Intervall of iteration could be 1/100 second or 1/1000 second what ever is possible.  Only result of this loop is to recognize  signal has changed or not, by comparing last signal value with current signal value. So this loop has true or false as result. And if it's false three times in series - we assume master won't control the object longer and therefore use pupil's signal. Every parameter (time intervall) , count of iteration in series (here three times) must be figured out from your side. And it could be wise to have different paramter (especially count of iteration) to overtake control to the master. Every blink of an eye could save money :-).  
Because I'm not familiar with signal's stability if nothing happened at a stick - is it really same value all the time? Assuming we have a signal value 1340 - does it range a bit? For example in between 1338 to 1342 or is it 1340 every time?

Woita2095, 2018-06-18:
There are two different point of wiew:
1) switching control among the master and the pupil via stick movement - there is disadvantage, when the master can't be sure whether the control is passed to the pupil already, or the master is still driving. So if there is such uncertainity, the master rather will take control over plane and this way the master would driving all the time. :-)
2) If a switch is used for hand over the driving, it is more clear for the master - he is sure who is driving. But the switching by the switch costs some time.

scvhorus12, 2018-06-18:
I still prefer option 2  :-).
Looking at option 1 - out of my view it's clear - master leave sticks untouched and in that case pupil's work and control the object. Switch option costs time and worts case money as well. That's why I'm struggeling with that option. 
Option 2 would be so smart and comfortable.

Woita2095, 2018-06-18:
Best way is to have boths options, the user can decide which one to use.

scvhorus12, 2018-06-19:
I have to corret formerly statement - I prefer solution without switch :-) - so it's option 1. And option 1 is smart and comfortable.  
Implementing both variants would be great - of course. Maybe I'm wrong, but I guess that 98% will use the solution without switch.

Woita2095, 2018-06-19:
OK, I have understand ;-) It was clear from context.