FrSky-OS/HORUS-X10-X10S-X12S-bug-report #203: All commands freezed et data log very strange

Michel79, 2018-05-28:
Hello.
Yesterday, I've crashed a kerosene turbine Jet and I think the reason is a bugg of FrTx V 1.5.05. 
I've lost the control when my model was less than 200 meters from me.
I used a Horus X12S and FrTx V1.5.05 FW. I used internal RF system with a X6R receiver EU LBT and an external module R9M/200mW NT with a R9 receiver EU LBT. I updated the firmware of Tx, RF module et Rx four days ago from FrSky site. The two RF systems simultanitly worked. In test range mode, the lost control distance was 50 m with the X6R and more than 180 m with the R9. 
I think the problem is a FrTx bugg because : 
After the crash and before turn off the transmiter, I've seen my model pic on the screen was replaced by the default pic "Horus X12S" and some special fonction like ReadBattery didn't work.
Now, after turn off and turn on again the transmiter, the model pic is displayed and ReadBatery work again
On the logs, some data are very strange : 
On the first flight, the name model "P51" is wrong ( name must be "Alpana", as for the others flights) and the year is "2017" instead "2018". But during this flight, I've not occured any problem.
On data of the last flight, J1,J2,J3,J4,PPM0,PPM1,PPM2,...values are not correct. For example, J1,J2 always are at zero. 
Included file : data log and model bin
[ALPANA.zip](https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/files/2046200/ALPANA.zip)

Regards, Michel.

Adelaxie, 2018-05-29:
Will look into this . sorry for your situations and will contact with you here if any progress.

strgaltdel, 2018-05-29:
Ich kehre zurück am 11.06.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht
"[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] All commands freezed et data log very strange (#203)"
gesendet am 28.05.2018 23:27:31.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

Adelaxie, 2018-05-30:
@Michel79  
1. Based on the log file , when is the crash happened?
2. P51 is the name of your other model?
3. How are the external and internal module works, seperately with two receiver or as backup?
 
Can you please leave your mail address? I may need to know more details about your model and crash later.

Adelaxie, 2018-05-30:
>On data of the last flight, J1,J2,J3,J4,PPM0,PPM1,PPM2,...values are not correct. For example, J1,J2 always are at zero.

You can see that the value is dislocation, not 0/

Michel79, 2018-05-30:
Hello. 

You can use my adress : michel.sole24@gmail.com <mailto:michel.sole24@gmail.com> .

 

- The crash happened at the last line of log file : 27/05/2018        17:18:22

- P51 is the name of an other model, but I've dont use this model at 27/05/2018. This day, I'ved only loaded "ALPANA" and I'ved never change.

- The two modul internal and external were both "on". The internal modul was bound with a X6R and the internal R9M was bound with a R9. In the model, The X6R and the R9 were linked by S-BUS on "Rx1 in" and "RX2 in" on the Redundancy Bus RB20. I used channel 1 to 12.

- Telemetry was only on the X6R. Its Smart port was plugged on "Rx1 S.P. "on the RB20.

- About J1, J2 values, I don't understand : "You can see that the value is dislocation, not 0/"

 

Michel.

strgaltdel, 2018-05-30:
Ich kehre zurück am 11.06.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] All commands freezed and data log very strange (#203)"
gesendet am 30.05.2018 10:04:40.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

Michel79, 2018-05-30:
Oh yes, I understand, now.

 

I see :  the logs have 8 columns added before J1,J2,… : celSum   celsMin rb1Volt rb1Amp               rb1Consum        rb2Volt                rb2Amp               rb2Consum

 

Michel.

strgaltdel, 2018-05-30:
Ich kehre zurück am 11.06.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] All commands freezed and data log very strange (#203)"
gesendet am 30.05.2018 12:32:52.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

Michel79, 2018-05-30:
Hello.
I think I've found the explaination of P51 data : these data have been recorded last year, when I flew with this model. The ALPANA data have been added in the same file last week because there no year in the file name.

strgaltdel, 2018-05-30:
Ich kehre zurück am 11.06.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re:
[FrSky-OS/HORUS-X10-X10S-X12S-bug-report] All commands freezed and data log very strange (#203)"
gesendet am 30.05.2018 13:54:19.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

Adelaxie, 2018-05-31:
@Michel79 Shock！ We discussed about the log file.if its created last year it may happen this. but we thought its impossible for you .Thanks for your further information.
First of all. the log file has errors that we need to fix it.
Secondly. I think there was reboot happened during the flight.
We will try to reproduce the reboot to see if data will have chance to loaded wrong values.

Michel79, 2018-07-25:
Hello.
I've proceeded to many test simulation, but I've never reproduce the problem.
But, I've modified the log file to eliminate data from the last year flight. Now, the data seems correct. Let's see included file.
[Vols 27-05-18 (crash).xlsx](https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/files/2227345/Vols.27-05-18.crash.xlsx)

Michel79, 2018-07-25:
But the fact still remain is recording has ended when commands has freezed.