FrSky-OS/HORUS-X10-X10S-X12S-bug-report #209: Firmware upgrade from 2017-03-22   (1.3.01)  to  2018-05-07     (1.5.05)

ixopa, 2018-07-05:
Hardware Number code  is :X12SEU14 and release firmware is for the moment stopped to 1.5.04 

I'm the owner of a HORUS X12S bought last week in Netherland via internet. Discovering this "beautifull" radio, I have read carefully the manual and start a full charge of the battery first. After 10 hours waiting that the green led should be off , I have fianally decided to upgrade the TX received from  1.3.01 to finally the 1.5.05 . All has run perfect until the latest step 1.5.05 . After rolling out the upgrade, I have start my radio and immediately the TX has begun to emit a sound that was near the "ultrasound". Extremely high at a point that it was uncomfortable ... I have immedialy put the key for sound "in quiet mode" and it was still there. Suddenly a smel of burning electronics" was coming from the HP vent ... I have immediately switch of  the radio hoping that he was still running. I have downgrade the radio to 1.3.01 and then restart to upgrade to 2018-04-09     1.5.04 the version just before. I'm afraid to restart the upgrade to the latest release. This can conduct on a damaged TX . Is FRSKY aware about this issue ? My TX is well functionning and all seems to be normal. Lucky that I have switch the TX immediately when I have smel the electronics odor.  Hardware Number code  is :X12SEU14 and release firmware is 1.5.04

Adelaxie, 2018-07-06:
Hello, @ixopa ,if the radio works well now after you downgrading it to 1.3.01?
We have not received any reports on the issue you mentioned. So, we need to know more details. thanks.

ixopa, 2018-07-10:
Hello. What can I said more. The only data's I can give you about the radio "batch" is : Number code is :X12SEU14 and release firmware is 1.5.04 for the moment. I have restarted the firmware upgrade from the version 1.3.01. The transmitter seems to work fine except a hardware problem already reported to service@rcstudio.cz regarding the charge of the battery that doesn't work properly. Shop having sell the radio is aware of the problem. Here, I'm in front of a software release and what I have explained is real. Upgrading to version 1.5.05 is risky again . The "ultrasonic" sound coming from the radio was going to burn the sound transducer. I was able to hear in some position the very high frequency  sound  coming out the radio. I would  like to continue the upgrade but my battery can't be charged. Problem is not the battery but de charging board and a request is placed to have a spare parts to service@rcstudio.cz they didn't answer. I'm a "Futaba" user from 50 years ago now and was really excited to use this HORUS and jump definitively to FRSKY. But with my bad experience up to now, I'm disapointed. Thanks

janekx, 2018-07-11:
I understand you have repported this at 7.7 at saturday, we proccess emails slowly as we have many emails daily and every one needs corresponding care. 
To your email > we cant send any part of the service center for self repair as you ask (charging board), as we must incpect other relationchips before and after repair, testing and we must keep company rules about repairs. In the email is not any sign about the speaker issue Btw.
Anyway is very strange that your dealer not care about the issue, this is what we not like to see.

ixopa, 2018-07-11:
Dear Sir
Thanks for your email that is clear . I understand your point of view. I have bought my radiovia internet in the Netherland market. My contact is good with the dealer. Looking to a lot of forums regarding the battery issue and particularly the site of "Alofthobbies" , it seems that thiscan happend with transmitters  distibutes earlier when released. I'm pretty sure that this radio wasone of the first of the shop. This is a personal point of view only. 
My approach is that I was in state to finally buy an HORUS and  I'm frustrated by a problem of battery and software that I seem to be the only one to have. I've been flying with Futaba for 50 years and never encounter such problems. I've been waiting for this radio for almost 3 years hoping to replace my Futaba T14SG with the HORUS. I know the FRSKY material because I use FRSKY FASST receivers. It is very reliable material. But I want efficient telemetry that HORUS allows me to have with my X8F receivers

I have also read that the FRSKY HORUS X12S suffer of a too less capacity regarding the time the TX is properly working. The nice color display sucks a lot of current.Some people have bought a set ( board and LIPO ) allowing them to increase working time and decrease the charging time. A question about : " is the TX still correctly working " with 7.2 V only and did this have not impact on the sending range ?Is this safe ? 
If the NIMh battery problem is resolved, what about the fact that I have to leave at least 2 year ( warranty time ) with a battery that seems from the beginning of the radio live on the market not really "adapted seemless" regarding workingtime on field. FRSKY is not the only TX suffering on too less capacity to sustain the current consumption ...
Thanks
Kind regards   
Francis Le Hardy

Woita2095, 2018-07-11:
Hello Ixopa,
I have heard about several incidents, when the radio had problem with the set of Lipo and charging module, even the radio was destroyed. Janekx could be able to say more about it.

Adelaxie, 2018-07-12:
@Woita2095 
>I have heard about several incidents, when the radio had problem with the set of Lipo and charging module, even the radio was destroyed.

Lipo can be used on X12S but the internal charger can not be used to charge a Lipo.
Am I understanding something wrong?

janekx, 2018-07-12:
@Adelaxie there is module with charger for lipo and battery, but using of unoriginal parts like not genuine charging board interrupt warranty.

ixopa, 2018-07-12:
Dear Adela
Thanks for info's. internal charger board of FRSKY Horus X12S is designed to charge NiMh battery only. Sure that using it is absolutely not recommended with LIPO due to risky failureof the LiPo that can burn. Charging LiPo , LiFe ... are based is completely different than NiMh and CdNi .Each cell of the LiPo need a self voltage protection to equilibrate itself regarding other cells to avoid overload , heat and a risk of battery failure. 
Thanks for your help. 
Kind regards   
Francis Le Hardy

ixopa, 2018-07-12:
Dear Sir
Thanks to pay attention on my issue. I'm glad to hear feedback about the charging board and LIPOset you can find on the market . Battery seems to be a big issue in the HORUS X12S. I can't understandthat a product expected to be released by a lot of people and is the top of the FRSKY products has beenlaunched with such an issue.  
Having a nice day
Kind regards   
Francis Le Hardy

janekx, 2018-07-12:
There si no issue normally with the NiMH, if you have some issue it is case of the one product.

ixopa, 2018-07-12:
With all my respect , reading french and american forum's this seems not really the case. I suggest to close here the discussion about this subject. To keep waranty , I have decided to contact the dealer in Holland and start a replacement or a repair procedure. I'm in front of too many problems with a full new acquisition I never had the time to play with. I'm still in the phase of exploring the product and continuously stopped by a battery failure . Sure that I'm frustrated. Many thanks for your support and comments. Have nice day. Kind regards

closed 2018-07-12 by ixopa

Woita2095, 2018-07-12:
Hi ixopa, in the mentioned set there is a charging board to replace instead of the the original board (little board next to battery). Then it is possible to charge by\ the internal charger.

> ixopa: Lipo can be used on X12S but the internal charger can not be used to charge a Lipo. Am I understanding something wrong?

Adelaxie, 2018-07-13:
thanks @Woita2095

ixopa, 2018-07-13:
Dear Sir
Thanks for your support and advise in that matter. 
Sorry that I have "closed" the subject that is still pending due to a false manipulation.

As read on Github , I have decided to send the radio back to Holland. I have some fear with the one I have in my hand today after problems encountered during discovery. I'm convinced that it is a well made radio. 
Kind regards    
Francis Le Hardy