FrSky-OS/HORUS-X10-X10S-X12S-bug-report #212: Special Function menu ( 6.1.11 )  ISSUE : "play track" doesn't play sound even wave file exists !

ixopa, 2018-08-03:
Full new HORUS X12S   hardware X12SEU14  Software 1.5.05  PhysID   23

I have a Eagle 3 flight stabilizator giving the ability to be setup in a choosen mode flight like GYRO Off, Normal Mode ( GYRO On ) and Rescue  Mode for example . To do this, you assign a switch with 3 positions on the TX. Assigning works well. I have assigned switch C to set the 3 mode of fly. Having done this, why not to trigger a voice that clearly mention the state you choose on flight. I have than create 3 voices for each situation  I have labelled as Gyro Of.wav, Gyro On.wav, Rescue.wav. The sounds waves files doesn't exceed 100k , are *.wav format and doesn't exceed 10 char. I have saved all the files in the folder "Track" under SOUNDS\en\ . Having done this, In the "Special Function" tab, I have check SF 1, set second column to "once", third to the switch SW C _Up , then fourth "Play track" and fifth select the "Rescue. wav" as file to play. I have repeated this process to set up the second and third position of the SC swith to play Gyro Of and Gyro On waves. I have then tried ( NOTHING comes out of the speaker ) .Being in front of that issue, I have simply copy one of the default waves from value to "Track" folder to see if the problem was my files. NO . Same result selecting that file instead of mine. The procees don't play the record. Afterwards, I have checked if the issue I was in front of was also a problem when reading a value instead of "Play a Track" . It works perfect . Conclusion, it seems that an issue exists in the HORUS software when you want to play a track triggered by a switch .  

Thanks to pay attention to this issue

ixopa, 2018-08-07:
To complete my post about "Special Function menu ( 6.1.11 ) ISSUE : "play track" doesn't play sound even wave file exists" , the excel file "sounds list.xls" containing name of sounds for System, Value and or tracks has been modified to locate where tracks can be found : SOUNDS/en/Track/cut.wav       cut.wav for example

ixopa, 2018-08-07:
Click on the wrong button. Issue must stay open . Thks

MIKE-DAILEY, 2018-08-07:
Try the attached and known good wav file.  Unzip it and put it on your flash memory drive (not the SD card) in SOUNDS/en/Tracks folder. Them in Special Functions select it for Play Track for a switch position.  If it can be seen and selected then test if it will play with the switch used.  If will not play check your MDL/SOUNDS screen to make sure your sound settings are at normal volume and not muted.  If it then does not play you will want to contact your FrSky dealer for repair of your system.  If it does play then check the wav file specs in the "how to" for your system on the FrSly download website.   

[RateHigh.zip](https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/files/2267076/RateHigh.zip)