FrSky-OS/HORUS-X10-X10S-X12S-bug-report #213: Spontaneous Shutdown

choochoo2, 2018-09-10:
I just had a very scary occurrence.  In mid session, the X10s just shut down.  The battery is freshly charged and read full both before and after the incident.  While shut down, the only thing operating was the blue light on the power button which didn't go out until I shut it off, no display, no transmission.  It went down once and then apparently rebooted, then it went down again permanently.  At the time, I was rapidly toggling modes for a flight controller programmed to ch 5 and the LS.  The firmware is FrOS 1.2.02 NEU.

VERY fortunately this happened in my garage while setting up a plane so the only harm was to my confidence.  Now, flying with this tx is a bet-your-plane proposition that scares me.

janekx, 2018-09-11:
Hi, 1.2.02 is very old FW, Now is 1.3.01 so we reccomend to update FW to be sure there is no bug you can meet.

choochoo2, 2018-09-11:
Is there a reason to believe this bug has been corrected in the newer firmware or are you just waiting until it shows up again?

choochoo2, 2018-09-11:
Sorry, this was closed accidentally.

janekx, 2018-09-11:
There was some bugs that result wired behave when you use some programming functions.