FrSky-OS/HORUS-X10-X10S-X12S-bug-report #214: LUA script documentation

eijowk, 2018-10-04:
I guess that there are many of us that are prepared to help out in rewriting or develop new LUA scripts for FROS 1.6.0.1 and higher.
But we can not guess...please provide some simple documentation and/or a few examples of working scripts. Then we can start develop and share!

goryc, 2019-03-03:
Also for me it would be important to know how the LUA scripts work: I can execute them and then I do not know how to use them. A documentation to use them would be necessary. Thank you.

Woita2095, 2019-03-03:
Some basics:
https://www.google.com/search?q=lua+scripts+how+to&ie=utf-8&oe=utf-8&client=firefox-b
;-)

goryc, 2019-03-04:
I thank you so much; I have already taken a look at this documentation and soon I will try to write something; my request was more trivial, I can not run for example "gtrans_fros.luac" present in the latest firmware version 1.6.03 beta. I start the application, change the channel, the power of the vtx but I do not get any result; I would like some documentation on the use of these scripts:
1. gaSuit_fros
2. gtrans_fros
3. sbec_fros
4. SXR_fros
I hope I explained it better.
Thank you

goryc, 2019-04-14:
News? I would like to use the script for VTX but it doesn't work!