FrSky-OS/HORUS-X10-X10S-X12S-bug-report #215: Horus 10s hung and I had to turn it off and on again

Derry061, 2018-10-10:
I was playing with my brand new 10s and using the 'SXR Setup' on the SYS menu. I had a S*R rx bound to the tx. I was switching back and forth to screen 2, trying to figure out the settings (I cant find any documentation giving details). Suddenly everything froze. None of the controls would respond. I had to turn off the Tx and turn it on again to get it to work. I am using version 1.2.05 which seems to be the latest full release.

Adelaxie, 2019-02-28:
Please help to confirm that we have fixed it on latest release. you can download the firmware from DOWNLOAD section of FrSKY website.
thanks in advance.

Derry061, 2019-03-07:
I have installed Frsky OS v 1.3.03 Beta and tried to reproduce the problem without success. so I guess that it is fixed.
Derry061

Adelaxie, 2019-03-11:
Thanks for your report.

closed 2019-03-11 by Adelaxie