FrSky-OS/HORUS-X10-X10S-X12S-bug-report #219: SXR Setup menu not working

choochoo2, 2018-10-22:
Actually this menu is a bit of a mess.  It doesn't seem usable to setup a S6R, it has an obvious bug, and the labeling on menu items is unnecessarily confusing.

To keep this post short, there is an attached pdf with detailed information explaining the problems and offering some suggested improvements.

[Problems with the SXR Setup menu in X10 FrOS V1.pdf](https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/files/2503419/Problems.with.the.SXR.Setup.menu.in.X10.FrOS.V1.pdf)

choochoo2, 2018-12-28:
This problem seems to be corrected in the X10 firmware release 1.3.01.  The menu now correctly displays an option to select quick-mode.  I am particularly happy to see the labeling has been improved and is much clearer now.  Thank you.

The SXR Lua script still does not function in the X10 with FrOs but since the embedded menu is working there is no apparent need to use the Lua.

choochoo2, 2019-01-11:
Unfortunately there is still a problem with the SXR setup menu.  The selection "polarity" is reversed for the ch5-6 functions.  For example; the label for ch 5 says "Enable ch5 as Ail2" but selecting "Enable" in the drop down box actually disables Ail2 and ch5 functions as an auxiliary pass-through.

I feel partly responsible for this problem as I suggested the labels you used.  When doing so I assumed you would adjust the selection in the drop down box to match but should have said this was needed in the original posting above.  My apology.

The simplest fix would be to reverse the sensing of the selection box so that selecting "Enable" enables Ail2 on ch5 (Same for Ele2 on ch6).

A better fix would be to change the label to "Select ch5 function" and change the text choices in the selection box to "AIL 2" and "Aux 1".  Ch6 would say "Select ch6 function" with the choices "Ele 2" and "Aux 2".  This would make it even more clear what functionality is being chosen.  Of course the function must match the text.

Adelaxie, 2019-01-14:
[S6Rconfig.zip](https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/files/2753873/S6Rconfig.zip)
Copy directly to flash/config

choochoo2, 2019-01-14:
Perfect!

Adelaxie, 2019-01-15:
resolved？ @choochoo2

choochoo2, 2019-01-15:
The issue in question is resolved in the corrected file in your post above, yes, but the problem still exists in the 1.3.01 firmware version posted on the website for everyone not following GitHub.  And please don't just plug this corrected S6Rconfig file into the website download.  That would create yet a third version of the firmware with the same version number.  In fact, in the future please issue a new version number when posting any version that is different in any way from a previously posted version.

Adelaxie, 2019-01-17:
Will do as you suggested , thanks.

choochoo2, 2019-02-26:
V 1.3.03 contains the corrected config file so this problem seems to be fixed.

closed 2019-02-26 by choochoo2