FrSky-OS/HORUS-X10-X10S-X12S-bug-report #222: Firmware V1.6.01 EU Beta Dual/Rate mistake

isidresa, 2018-11-13:
Please see the following pdf document:
[frsky info dual_rate problem in firmware 1.6.01 beta.pdf](https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/files/2577550/frsky.info.dual_rate.problem.in.firmware.1.6.01.beta.pdf)

choochoo2, 2018-12-22:
I just discovered the same problem in X10s firmware 1.3.01 beta trying to adjust flap positions.  Exploring a bit further, this problem applies to all variants, line, multi-line, smooth, etc. including expo, where it will not accept negative numbers for any of the parameters.  This whole menu section seems to be buggy.

Adelaxie, 2018-12-24:
We will release normal version of v1.6.1 for x12s / v1.3.1 for x10/s which fix some bugs like this in v1.6.1beta / 1.3.1beta. Thanks for your reports.

choochoo2, 2018-12-26:
This problem has been corrected in X10 1.3.01 release version.  Negative values are now working as expected.

Adelaxie, 2018-12-27:
@choochoo2  Thanks for the confirmation.

closed 2018-12-27 by Adelaxie