FrSky-OS/HORUS-X10-X10S-X12S-bug-report #224: Problem with Timer in X10 1.3.01 release

choochoo2, 2018-12-26:
Timer 1 is setup as TH%.  It should start counting only when the throttle is advanced and count at a rate proportional to the throttle.  It did this in 1.3.01 beta.  In the new release version the timer starts counting immediately when the system is powered up and counts at a rate that is about 5x full speed.  That is to say it speeds through a 5 min countdown in about 1 min, even without advancing the throttle.

Adelaxie, 2018-12-27:
@choochoo2  Have not reproduced of it. any chance of a video? thanks in advance.

choochoo2, 2018-12-27:
Here is a video illustrating the problem.  It isn't great, there is some difficulty with reflections and focus shifting but I think you can see what you need to see.  If necessary I will re-make it.

Some additional information:  Timer 1 is running at hyper speed as long as the throttle is off.  It continues if the throttle-cut is disengaged but returns to normal behavior if the throttle is advanced.  Timer 2 runs at real-time but only when throttle-cut is disengaged.  I suspect the problem may also be connected to the fact that throttle channel travel goes to -125%.

https://youtu.be/XJ9gYqxECeQ

Woita2095, 2018-12-28:
Choochoo, try to have a look to Channel Monitor, especially to throttle curve. Have you used any throttle curve?

choochoo2, 2018-12-28:
The throttle curve is at the defaults, it has never been used.  Also, when the plane was flown last week, the timer was working normally with 1.3.01 beta firmware.  On the 26th I installed the new 1.3.01 release firmware.  When the transmitter was booted after the installation the runaway timer occurred.  Nothing in the timer or throttle was changed between the flying and the firmware installation, in fact I don't think the transmitter was turned on (holidays).

The throttle travel on the model that is misbehaving is set to ±125%.  When the range is set to ±100% the throttle behaves normally.  The problem seems to occur when the throttle is lower than -100%.  If you are trying to duplicate this problem, first set the throttle lower travel to -125%.

Adelaxie, 2018-12-29:
>The throttle travel on the model that is misbehaving is set to ±125%. When the range is set to ±100% the throttle behaves normally. The problem seems to occur when the throttle is lower than -100%. If you are trying to duplicate this problem, first set the throttle lower travel to -125%.
Will try this. thanks
@Woita2095 thanks for help.

choochoo2, 2019-02-26:
This appears to be fixed in 1.3.03 Beta.  Will re-open if further problems are discovered.

closed 2019-02-26 by choochoo2