FrSky-OS/HORUS-X10-X10S-X12S-bug-report #225: Annoying problem with Switch Warning

choochoo2, 2018-12-28:
I have been noticing a minor but annoying problem with the Switch-Warning feature.  Not sure what triggers it, making it hard to reproduce, but sometimes the warning screen pops up even when the switches are in the correct position.  Moving the switches at this point will not dismiss the warning, it only changes the switch color on the display.  To dismiss the screen one must click enter.  Once this problem starts happening it seems to persist.  It can be corrected by going to the Switch Warning screen, changing nothing, and executing a long press of enter to re-save the already existing settings.

This has occurred in 2-3 different models using 1.3.01 beta and is still happening with 1.3.01 release.

Attached is a screen shot of the warning occurring even though both switches are in the correct position.

[image]

choochoo2, 2018-12-28:
Probably nothing can be done about this problem unless someone identifies what triggers the problem so it can be replicated.  It should be mentioned that this is not new to 1.3.01 release, it occurred also with 1.3.01 beta, not sure about earlier.

Note that it should not be possible to capture the screen shot shown above if things are working normally.  This screen only appears when the system boots with at least one switch in the wrong position.  The switch in the wrong position is shown in red while the correct switches are in white.  The instant that all switches are moved to the correct position, and therefore would be white, this screen closes and the boot commences.  Having all switches in the correct position and shown in white as is the case above should never exist for more than perhaps a small fraction of a second.  When this problem occurs it boots to this screen as shown and persists until ENTER is pressed to skip, regardless of switch positions.

choochoo2, 2019-04-20:
This problem has recurred today with 1.3.03.  It is still happening now, so let me know if there is anything you would like me to try.  It is not cleared by powering off the tx or changing models.  It is cleared by using the SWITCH WARNING screen with a long press to set the warning states, even if nothing is changed.

MIKE-DAILEY, 2019-04-20:
Does it always show the same switches?

Mike

choochoo2, 2019-04-20:
It's the same switch during the same incident but I don't think it's always the same switch for all incidents.  Compare the current screenshot (attached) to the earlier one.  Both are warning SG but the earlier one also references SE.  The two incidents are different models and model types.  On both, SG is a throttle cut but I suspect that switch is showing up simply because it isn't disabled on the SWITCH WARNING page.
[image]

MIKE-DAILEY, 2019-04-20:
I've never had that happen.  I wonder if it would be worthwhile to disconnect the switch plugs where they connect to the main board then reconnect and reseat them.

choochoo2, 2019-04-20:
Apparently it's only happened to me but it has happened at least three times, in different model memories, model types, and with different firmware.

It's sensing the switch correctly.  When the switch is in the "safe" position it shows in white.  In other positions it shows in red.  This will change if the switch is flipped but the warning screen persists regardless of switch position.  The switch works correctly in other models, and in this model after the warning has been bypassed.

Some added background; Prior to this current incident this model, named "Quad" had not been flown for months and presumably the model memory had not been selected.   When Quad was selected the bogus warning message appeared.  The warning appears whether turning the tx on, or changing to Quad from a different memory.

MIKE-DAILEY, 2019-04-20:
Very strange problem.

Lombardoa, 2019-05-16:
Hi, in the past I had the same problem. Everything happened after doing the calibration of the sticks (STICK CAL). in fact, every time I changed a model, it showed me the switch warning page.
I solved the problem by updating, again, the sw warning page, with the throttle stick slightly up, for example from -100 to -90. to avoid any fluctuations present at -100.