FrSky-OS/HORUS-X10-X10S-X12S-bug-report #226: Trims Lost on X10 1.3.01 release

choochoo2, 2019-01-02:
When changing model memories, the trim settings are not retained when returning to the first model.

Also a suggestion;  When releasing a firmware version that was previously posted as a beta, please assign a new version number unless the release version is **IDENTICAL** to the beta.  In a situation like we have now with 1.3.01 beta and 1.3.01 release not being the same, the words "beta" and "release" effectively become part of the version number as they are needed to identify which version is being referenced.  That's what the version number should be for.

MIKE-DAILEY, 2019-01-03:
Confirmed on 1.6.01 as well.

Adelaxie, 2019-01-11:
trim setting？ trim setup？please  clarify which parameter or which page?thanks.

choochoo2, 2019-01-11:
Lets say you fly airplane "A" and add 10 clicks of up-trim to obtain level flight and then change model memories to fly airplane "B", when you later return to airplane "A" the 10 clicks of trim you entered are gone and the trims are now back to neutral.  To fly airplane "A" again you must re-trim the plane.  You must re-trim each time you change aircraft.

Woita2095, 2019-01-11:
I confirm the same behavior at described procedure.
The trims are cleared everytime the models are switched.
Please fix it ASAP, it is very severe bug. The radio is not possible to use for flying.

Woita2095, 2019-01-11:
Confirmed the same behavior at the Horus X12s, FW 1.6.01L.

Adelaxie, 2019-01-15:
Please help to confirm if its fixed on 
http://down.frsky-rc.com:8001/Adlea/.x10Beta0115.rar
http://down.frsky-rc.com:8001/Adlea/.x12Beta0115.rar

choochoo2, 2019-01-15:
The link above fails on download attempts.

OK, now it works.

MIKE-DAILEY, 2019-01-15:
Adela,

I’m not able to download from your links, but apparently from the user post on Gethub the saving trim issue is fixed.

Any update on the THs and TH% counter and acquiring the vairo problem fix?

Mike

http://down.frsky-rc.com:8001/Adlea/.x10Beta0115.rar<https://eur02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fdown.frsky-rc.com%3A8001%2FAdlea%2F.x10Beta0115.rar&data=02%7C01%7C%7C902b944a66ac4e9ac8dc08d67abee3e4%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636831357609412471&sdata=E46%2Fnm8NcV14qnEOJYSA2dRFeG%2F2hl%2F%2BNMH1SDlmVhw%3D&reserved=0>
http://down.frsky-rc.com:8001/Adlea/.x12Beta0115.rar<https://eur02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fdown.frsky-rc.com%3A8001%2FAdlea%2F.x12Beta0115.rar&data=02%7C01%7C%7C902b944a66ac4e9ac8dc08d67abee3e4%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636831357609412471&sdata=l4%2FYYjjgAK6EhBhymAV8V7bI0NmSkcJf2KnTZ2V3aJA%3D&reserved=0>

choochoo2, 2019-01-15:
1.3.02 beta seems to be moving in the right direction:

- The trims are retained when changing models
- The problem with curves in expo/rates is corrected.
- The problem with timer TH% still exists (did not check THs)
- It still contains the older S6Rconfig file. It should get the newer, corrected, file posted recently on GitHub.

BazG2, 2019-02-24:
ok, think trims are ok now, cheers

choochoo2, 2019-02-26:
The problems with trims and curves were corrected in 1.3.02.  The problems with the timers were corrected in 1.3.03 Beta, which also contains the updated SXR config so all four problems bulleted above are now corrected in 1.3.03 Beta.

closed 2019-02-26 by choochoo2

MIKE-DAILEY, 2019-02-26:
Congratulations to the FrSky developers to getting the fixes.