FrSky-OS/HORUS-X10-X10S-X12S-bug-report #233: Possibility of the usage on FLEX Firmware at 868 mhz R9m AND EU or NEU at 2,4 Ghz internal Module

brainbubblersbest, 2019-03-03:
Is there no possibility of using The internal 2,4ghz RF Module at Flex Firmware?
Will use R9M Flex for long Range and 2,4 internal RF for  Race Quad.

At this time i have FLEX Firmware 1.6.01 on Horus x12s and Flex on R9M and FLEX F.PORT on all R9 Receivers.
But my internal RF is now Useless. Same with all my 2,4 Receivers, cause they wont bind anymore

Adelaxie, 2019-03-06:
Now on FLEX version of FrOS, the internal module will be in FCC mode. Did you have a try to use the internal module to work with receivers in FCC mode?

brainbubblersbest, 2019-04-08:
Yes i tried. But why is that not possible.
Using an eulbt 2,4ghz Transmitter and as an Option, use the R9m Flex System with the possibility to change the 900mhz to 868mhz and to adjust TX Power of a flex flashed r9 Module. In Germany if I use Flex i cannot use the2, 4ghzTX in the case i go to the rc club etc.

Adelaxie, 2019-04-24:
@brainbubblersbest  

[EU+FLEX_TESTING.zip](https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/files/3110815/EU%2BFLEX_TESTING.zip)

brainbubblersbest, 2019-05-14:
Hello, Ok lets give it a try. 
Are there any files to copy manually to the internal or SD card DRIVE? Because there is no readme. 
Or shuld i Do it same way as with another Version Number? Wich one? 
And if I testing this Firmware. Would this enhancement merged into future Firmwares?
Meaning usage of local legal internal 2,4Ghz tx at the RC Club and Modules for long range or other cases depends on installed Firmware on the Module?

I answer to this after next sunday after flashing and testing with a long range 7" quad at 868mhz and testing internal Module too. 

Thank you so far. Never expected such a fast response anyway. I'm already impressed.

Adelaxie, 2019-05-15:
Hi
If you are using the version which is old so that you would better to use the latest flash contents which is listed on website and then update to this test firmware.
if it works we can release this kind of version when new firmware release.
Please do bench test firstly as its not fully tested firmware.
You just need to check if the mode is correct and do a normal pre-flight checking.
Thanks and sorry for late reply last time. 
We have many tasks are under going so there is priorities to do it .

Thanks in advance, hope will get your test results later.

brainbubblersbest, 2019-06-01:
Hello, till today i have nothing negativ to report, everything work as it should. I Continue testing.
Will this be merged in to the ACCESS Update?

Adelaxie, 2019-06-05:
We will provide Flex firmware with LBT mode for internal module in future releases.
But its not related to ACCESS I think.